DJI-Mobile-SDK-Tutorials/Android-VideoStreamDecodingSample #1: Why the .so file is only support andorid 5.x ?

DennisGuo, 2016-10-21:
I can only run this example on android5.x or later . But there a lots of android4.x version devices. How Can  I support those ?
When load libraries on android4.x devices , the error shows below:
----------------------------------------------
`
AndroidRuntime: FATAL EXCEPTION: main

 java.lang.UnsatisfiedLinkError: dlopen failed: cannot locate symbol "atof" referenced by "libffmpeg.so"...

at java.lang.Runtime.loadLibrary(Runtime.java:365)

at java.lang.System.loadLibrary(System.java:526)
`

tito91, 2016-10-24:
Same error on Samsung Galaxy Note 4 (Android 4.4.4).

oliverou, 2016-10-26:
Hi, thanks for your feedbacks. We will investigate this issue in these few days and get back to you once we make progress.

lillogoal, 2017-01-22:
@oliverou please check here http://forum.dev.dji.com/forum.php?mod=viewthread&tid=33011&extra=page%3D1%26filter%3Dtypeid%26typeid%3D307 this is my issue

oliverou, 2017-01-22:
Hi @lillogoal, I think you have changed the package name of the nativehelper file, this kind of log is quite common, please check the following page in Google Documentation: <https://developer.android.com/training/articles/perf-jni.html>

[image]

lillogoal, 2017-01-22:
@oliverou Are you sure ? Maybe I have copied NativeHelper File and DJI.

NATIVE:
`package com.wesii.wesiiplay.Media;

/**
 *  A helper class to invoke native methods
 */
public class NativeHelper {

    public static final String TAG = NativeHelper.class.getSimpleName();

    public interface NativeDataListener {
        /**
         * Callback method for receiving the frame data from NativeHelper.
         * Note that this method will be invoke in framing thread, which means time consuming
         * processing should not in this thread, or the framing process will be blocked.
         * @param data
         * @param size
         * @param frameNum
         * @param isKeyFrame
         * @param width
         * @param height
         */
        void onDataRecv(byte[] data, int size, int frameNum, boolean isKeyFrame, int width, int height);
    }

    private NativeDataListener dataListener;

    public void setDataListener(NativeDataListener dataListener) {
        this.dataListener = dataListener;
    }
    //JNI

    /**
     * Test the ffmpeg.
     * @return
     */
    public native String codecinfotest();

    /**
     * Initialize the ffmpeg.
     * @return
     */
    public native boolean init();

    /**
     * Framing the raw data from camera
     * @param buf
     * @param size
     * @return
     */
    public native boolean parse(byte[] buf, int size);

    /**
     * Release the ffmpeg
     * @return
     */
    public native boolean release();

    static{

        System.loadLibrary("ffmpeg");
        System.loadLibrary("djivideojni");
    }

    private static NativeHelper instance;

    public static NativeHelper getInstance() {
        if (instance == null) {
[... 677 lines cut ...]
                codec.stop();
                codec.release();
                NativeHelper.getInstance().release();
            }

            @Override
            public void onDataRecv(byte[] data, int size, int frameNum, boolean isKeyFrame, int width, int height) {
                if (dataHandler == null || dataHandlerThread == null || !dataHandlerThread.isAlive()) {
                    return;
                }
                if (data.length != size) {
                    loge( "recv data size: " + size + ", data lenght: " + data.length);
                } else {
                    logd( "recv data size: " + size + ", frameNum: "+frameNum+", isKeyframe: "+isKeyFrame+"," +
                            " width: "+width+", height: " + height);
                    currentTime = System.currentTimeMillis();
                    frameIndex ++;
                    DJIFrame newFrame = new DJIFrame(data, size, currentTime, currentTime, isKeyFrame,
                            frameNum, frameIndex, width, height);
                    dataHandler.obtainMessage(MSG_FRAME_QUEUE_IN, newFrame).sendToTarget();

                }
            }
        }
`

oliverou, 2017-01-22:
Hi @lillogoal, you have changed the package names of the above two files:

[image]
[image]

You should change the corresponding method names in the **jni/dji_video_jni.c** file and rebuild the so libs.

Furthermore, you can check this StackOverFlow: <http://stackoverflow.com/questions/16518490/java-lang-unsatisfiedlinkerror-jni>

lillogoal, 2017-01-22:
so if i implements all your sample project ?

lillogoal, 2017-01-22:
I have this problem with re-compiler c++ files.

NDK BUILD.
`ndk-build
[armeabi-v7a] Compile thumb  : djivideojni <= dji_video_jni.c
jni/dji_video_jni.c:65:51: warning: passing 'uint8_t *' (aka 'unsigned char *') to parameter of type 'const jbyte *' (aka 'const signed char *') converts between pointers to integer types with different sign [-Wpointer-sign]
        (*env)->SetByteArrayRegion(env, jarray, 0, size, buf);
                                                         ^~~
1 warning generated.
[armeabi-v7a] SharedLibrary  : libdjivideojni.so
D:/Users/Lorenzo/AppData/Local/Android/sdk/ndk-bundle/build//../toolchains/arm-linux-androideabi-4.9/prebuilt/windows-x86_64/lib/gcc/arm-linux-androideabi/4.9.x/../../../../arm-linux-androideabi/bin\ld: error: ./obj/local/armeabi-v7a/libffmpeg.so:1:9: syntax er
ror, unexpected STRING
D:/Users/Lorenzo/AppData/Local/Android/sdk/ndk-bundle/build//../toolchains/arm-linux-androideabi-4.9/prebuilt/windows-x86_64/lib/gcc/arm-linux-androideabi/4.9.x/../../../../arm-linux-androideabi/bin\ld: error: ./obj/local/armeabi-v7a/libffmpeg.so: not an object
 or archive
jni/dji_video_jni.c:43: error: undefined reference to 'av_register_all'
jni/dji_video_jni.c:45: error: undefined reference to 'av_codec_next'
jni/dji_video_jni.c:78: error: undefined reference to 'avcodec_register_all'
jni/dji_video_jni.c:79: error: undefined reference to 'av_register_all'
jni/dji_video_jni.c:82: error: undefined reference to 'avcodec_find_decoder'
jni/dji_video_jni.c:83: error: undefined reference to 'avcodec_alloc_context3'
jni/dji_video_jni.c:84: error: undefined reference to 'av_parser_init'
jni/dji_video_jni.c:97: error: undefined reference to 'avcodec_open2'
jni/dji_video_jni.c:103: error: undefined reference to 'av_frame_alloc'
jni/dji_video_jni.c:127: error: undefined reference to 'av_init_packet'
jni/dji_video_jni.c:132: error: undefined reference to 'av_parser_parse2'
jni/dji_video_jni.c:165: error: undefined reference to 'av_free_packet'
jni/dji_video_jni.c:223: error: undefined reference to 'avcodec_close'
jni/dji_video_jni.c:227: error: undefined reference to 'av_free'
jni/dji_video_jni.c:228: error: undefined reference to 'av_free'
jni/dji_video_jni.c:229: error: undefined reference to 'av_parser_close'
clang++.exe: error: linker command failed with exit code 1 (use -v to see invocation)
make: *** [obj/local/armeabi-v7a/libdjivideojni.so] Error 1
`

pisarik, 2017-02-14:
Hello!

I successfully builded this sample. And it works fine on API >=21. Then this solution was embedded in my project. I replaced all neccessary methods in `dji_video_jni.c` with my packages. And my project working fine on API >= 21. When I trying to run it on my phone with Android API level 19, then i getting the following error:

```
FATAL EXCEPTION: main
Process: uiip.dji.pcapi.com, PID: 8922
java.lang.UnsatisfiedLinkError: dlopen failed: cannot locate symbol "atof" referenced by "libffmpeg.so"...
at java.lang.Runtime.loadLibrary(Runtime.java:364)
at java.lang.System.loadLibrary(System.java:526)
at uiip.dji.pcapi.com.media.NativeHelper.<clinit>(NativeHelper.java:65)
at uiip.dji.pcapi.com.MainActivity.onCreate(MainActivity.java:21)
at android.app.Activity.performCreate(Activity.java:5275)
at android.app.Instrumentation.callActivityOnCreate(Instrumentation.java:1087)
at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2166)
at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:2252)
at android.app.ActivityThread.access$800(ActivityThread.java:139)
at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1200)
at android.os.Handler.dispatchMessage(Handler.java:102)
at android.os.Looper.loop(Looper.java:136)
at android.app.ActivityThread.main(ActivityThread.java:5103)
at java.lang.reflect.Method.invokeNative(Native Method)
```

Foremost, I tried to set `targetSdkVersion` to 19 in `Build.Gradle (Module: app)` and add the line `APP_PLATFORM := android-19` in `libs/Application.mk` and rebuild jni libs. But this didn't help.

Then I found this [post](https://answers.unrealengine.com/questions/399263/why-ndk-android-21-or-later-wont-run-on-pre-50-dev.html). And decided to try to change `compileSdkVersion` to 19, I deleted all api-specific stuff as styles `appCompat` and also I made the method `decodeFrame()` in `DJIVideoStreamDecoder` empty (just as mock), because it contains method `mediacodec.reset()`, which is 21 level api. So it didn't help too.

Then I tried to compile with different older ndk versions (9b, 10e, 11c), and different Android Plugin Version (2.1.3). It didn't help.

Is it problem in libffmpeg.so? How should i deal with it?

talobin, 2017-10-17:
Will go ahead and close this ticket as current version should have a fix for this already.
Please feel free to reopen if the problem still exists.
Thanks!

dji-dev, 2017-10-17:
Will go ahead and close this ticket as current version should have a fix for this already.
Please feel free to reopen if the problem still exists.
Thanks!

closed 2017-10-17 by dji-dev