DJI-Mobile-SDK-Tutorials/Android-VideoStreamDecodingSample #3: Repo fails to clone

acortelyou, 2017-01-12:
When cloning this repo I get an error during the `git lfs fetch` phase.

> **$ git clone git@github.com:DJI-Mobile-SDK-Tutorials/Android-VideoStreamDecodingSample.git**
>
>Cloning into 'Android-VideoStreamDecodingSample'...
>remote: Counting objects: 480, done.
>remote: Compressing objects: 100% (323/323), done.
>remote: Total 480 (delta 112), reused 480 (delta 112), pack-reused 0
>Receiving objects: 100% (480/480), 54.56 MiB | 6.95 MiB/s, done.
>Resolving deltas: 100% (112/112), done.
>Checking connectivity... done.
>Downloading android-videostreamdecodingsample/jni/lib/libavcodec.a (184.02 MB)
>
> **Error downloading object: android-videostreamdecodingsample/jni/lib/libavcodec.a**
> **This repository is over its data quota. Purchase more data packs to restore access.**
> **https://help.github.com/articles/purchasing-additional-storage-and-bandwidth-for-an-organization/** 
> **Git LFS: (0 of 3 files) 0 B / 346.37 MB**

The project cannot be built without those three missing files.

Is there another source for these files?

How can I get this repo in its entirety?

dji-dev, 2017-01-12:
Hi @acortelyou , we just purchased the data packs, you should be available to download the file using git lfs now. Please try.

acortelyou, 2017-01-12:
Thanks @dji-dev, it is working now.