DJI-Mobile-SDK-Tutorials/Android-VideoStreamDecodingSample #5: Re-Compile JNI DJIVIDEOSTREAMDECODE

lillogoal, 2017-01-23:
Hello to all,
I'm trying to re-compile the source C ++, using the latest Android NDK builder.
The code is posted here:
https://github.com/DJI-Mobile-SDK-Tutorials/Android-VideoStreamDecodingSample/tree/master/android-videostreamdecodingsample/jni

I have this problem, this is the log:

[armeabi-v7a] Compile thumb  : djivideojni <= dji_video_jni.c
jni/dji_video_jni.c:65:51: warning: passing 'uint8_t *' (aka 'unsigned char *') to parameter of type 'const jbyte *' (aka 'const signed char *') converts between pointers to integer types with different sign [-Wpointer-sign]
        (*env)->SetByteArrayRegion(env, jarray, 0, size, buf);
                                                         ^~~
1 warning generated.
[armeabi-v7a] Prebuilt       : libffmpeg.so <= jni/
[armeabi-v7a] SharedLibrary  : libdjivideojni.so
D:/Users/Lorenzo/AppData/Local/Android/sdk/ndk-bundle/build//../toolchains/arm-linux-androideabi-4.9/prebuilt/windows-x86_64/lib/gcc/arm-linux-androideabi/4.9.x/../../../../arm-linux-androideabi/bin\ld: error: ./obj/local/armeabi-v7a/libffmpeg.so:1:9: syntax er
ror, unexpected STRING
D:/Users/Lorenzo/AppData/Local/Android/sdk/ndk-bundle/build//../toolchains/arm-linux-androideabi-4.9/prebuilt/windows-x86_64/lib/gcc/arm-linux-androideabi/4.9.x/../../../../arm-linux-androideabi/bin\ld: error: ./obj/local/armeabi-v7a/libffmpeg.so: not an object
or archive
jni/dji_video_jni.c:43: error: undefined reference to 'av_register_all'
jni/dji_video_jni.c:45: error: undefined reference to 'av_codec_next'
jni/dji_video_jni.c:78: error: undefined reference to 'avcodec_register_all'
jni/dji_video_jni.c:79: error: undefined reference to 'av_register_all'
jni/dji_video_jni.c:82: error: undefined reference to 'avcodec_find_decoder'
jni/dji_video_jni.c:83: error: undefined reference to 'avcodec_alloc_context3'
jni/dji_video_jni.c:84: error: undefined reference to 'av_parser_init'
jni/dji_video_jni.c:97: error: undefined reference to 'avcodec_open2'
jni/dji_video_jni.c:103: error: undefined reference to 'av_frame_alloc'
jni/dji_video_jni.c:127: error: undefined reference to 'av_init_packet'
jni/dji_video_jni.c:132: error: undefined reference to 'av_parser_parse2'
jni/dji_video_jni.c:165: error: undefined reference to 'av_free_packet'
jni/dji_video_jni.c:223: error: undefined reference to 'avcodec_close'
jni/dji_video_jni.c:227: error: undefined reference to 'av_free'
jni/dji_video_jni.c:228: error: undefined reference to 'av_free'
jni/dji_video_jni.c:229: error: undefined reference to 'av_parser_close'
clang++.exe: error: linker command failed with exit code 1 (use -v to see invocation)
make: *** [obj/local/armeabi-v7a/libdjivideojni.so] Error 1

pisarik, 2017-01-25:
I have stepped into the same problem and found a solution.
This happened, because I didn't notice in readme the important [section](https://github.com/DJI-Mobile-SDK-Tutorials/Android-VideoStreamDecodingSample#important-install-git-lfs). 

I noticed that libffmpeg.so takes only 1KB of memory. Meanwhile original libffmpeg.so takes about 60MB of memory and therefore it stored with help of [special module](https://help.github.com/articles/versioning-large-files/) . That is why your linking failed.

In that case you can complete linking without dipping into git lfs (but I am sure, that it is pretty simple).
You can just download this original [libffmpeg.so](https://github.com/DJI-Mobile-SDK-Tutorials/Android-VideoStreamDecodingSample/blob/master/android-videostreamdecodingsample/jni/libffmpeg.so) and replace the "placeholder" libffmpeg.so which located in app/jni

lillogoal, 2017-01-25:
@pisarik thank you so much :)

giannisergente, 2017-02-22:
Hi @lillogoal, we have the same issue (https://github.com/DJI-Mobile-SDK-Tutorials/Android-VideoStreamDecodingSample/issues/12). Did you solve also for arm64-v8a e x86 ABI? Because in those cases compilation fail :(

tubipham, 2017-04-09:
hi @pisarik i changed package name in dji_video_jni.c but i got some error:
rocess: com.sample.insightus.sample, PID: 25786
                                                                             java.lang.UnsatisfiedLinkError: No implementation found for boolean com.sample.insightus.sample.media.NativeHelper.init() (tried Java_com_sample_insightus_sample_media_NativeHelper_init and Java_com_sample_insightus_sample_media_NativeHelper_init__)
                                                                                 at com.sample.insightus.sample.media.NativeHelper.init(Native Method)

dji-dev, 2017-10-17:
Will go ahead and close this ticket as current version should have a fix for v8a and x86 already.
Please feel free to reopen if the problem still exists.
Thanks!

closed 2017-10-17 by dji-dev