DJI-Mobile-SDK-Tutorials/Android-VideoStreamDecodingSample #7: Which products can be supported by "Android-VideoStreamDecodingSample" project?

jamesxienctu, 2017-02-04:
Hi, all:
I run the project on my Android device and try to connect to DJI Phantom 3 Professional.
But the app shows "No Product Connected".

The message is as follows:
https://drive.google.com/open?id=0B726Quay7lRDUGVOc3hqb1NOY0U

My questions are as follows:
1. Which products can be supported by "Android-VideoStreamDecodingSample" project?

2. How can I modify this project to make it support DJI Phantom 3 Professional and DJI M100?

Please help answer them, thank you.

lillogoal, 2017-02-05:
Are you connected on WIFI of PHANTOM ?

jamesxienctu, 2017-02-05:
I run the project on my Android device, and my Android device is connected to remote controller by USB line.
Then remote controller is connected to DJI Phantom 3 Professional.

dji-dev, 2017-02-06:
Hi @jamesxienctu, the Android Video Stream Decoding Sample supports Phantom 3 Professional. The issue you mention may be due to the default app behavior for a USB Accessory settings of your Android phone. Please check the following FAQ to reset your Android system's default app settings and try again:

[How do I reset the default app behavior for a USB Accessory (DJI Product) on Android devices?](http://developer.dji.com/mobile-sdk/documentation/faq/index.html#how-do-i-reset-the-default-app-behavior-for-a-usb-accessory-dji-product-on-android-devices)

[How do I reset the default app behavior for a USB Accessory (DJI Product) on Samsung devices?](http://developer.dji.com/mobile-sdk/documentation/faq/index.html#how-do-i-reset-the-default-app-behavior-for-a-usb-accessory-dji-product-on-samsung-devices)

jamesxienctu, 2017-02-06:
I think it should not be the the default app behavior because I can connect to Phantom 3 Professional by other app.
https://drive.google.com/file/d/0B726Quay7lRDQlY3dU9ZUEVSYms/view?usp=sharing
https://drive.google.com/file/d/0B726Quay7lRDVUdIWmlNUHZYcWs/view?usp=sharing

But this app is strange. Even if it shows "Phantom 3 Professional Connected" underneath, I still can't  click Open button and status shows "No Product Connected"
https://drive.google.com/file/d/0B726Quay7lRDTmM3TXB5MzhBUkU/view?usp=sharing

Even if it connect to Phantom 3 Professional, after I click Open button, it shows error message.
https://drive.google.com/file/d/0B726Quay7lRDUDFiSFRaa0V2cEk/view?usp=sharing
https://drive.google.com/file/d/0B726Quay7lRDTmNpeUlrS1d4ZzA/view?usp=sharing

oliverou, 2017-02-06:
Hi @jamesxienctu, seems like you have a crash issue when you try this sample app. I have tested it the sample app on Phantom 3 Professional, it works well. To help you investigate the issue, could you send us the **crash log** of the sample app? And please provide the following infos for us the reproduce the issue:

-  Android system version you are using (Like Android 6.0.1, Android 5.1.1, etc)
-  Android device you are using (Like Samsung Galaxy Note 5, Nexus 5, etc)
-  Android Studio version you are using (Like Android Studio 2.2, etc)

jamesxienctu, 2017-02-06:
Thanks for your help, I have solved the crash issue.
But this app still has a problem with connecting to Phantom 3 Professional, I need to try several times to connect, and it has no crash log.
My Android system version is Android 5.1.
My Android device is Nexus 7.
My Android Studio version is 2.2.3.

dji-dev, 2017-10-17:
Will go ahead and close this ticket as current version should have a fix for this already.
Please feel free to reopen if the problem still exists.
Thanks!

closed 2017-10-17 by dji-dev