DJI-Mobile-SDK-Tutorials/Android-VideoStreamDecodingSample #10: Please upgrade libffmpeg.so, it's a old verson caused failed: dlopen failed.

niflying, 2017-02-20:
hi, I cann't run this sample. the error is:
failed: dlopen failed: cannot locate symbol "atof" referenced by "libffmpeg.so"

It loos like an old version there.

Please help.

oliverou, 2017-02-20:
Hi @niflying,  please follow the Readme file to clone the entire Sample Project using Git LFS and try to run the sample again.

niflying, 2017-02-20:
hi @oliverou, I am  follow the Readme file , use "git lfs clone https://github.com/DJI-Mobile-SDK-Tutorials/Android-VideoStreamDecodingSample" to got the project.  It's about 1GB , no files miss. 
But no lucy, It doesn't work too.

 E/dalvikvm: dlopen("/data/app-lib/com.dji.videostreamdecodingsample-2/libffmpeg.so") failed: dlopen failed: cannot locate symbol "atof" referenced by "libffmpeg.so"...  on    NativeHelper.getInstance().init();

niflying, 2017-02-20:
[android-videostreamdecodingsample-debug.zip](https://github.com/DJI-Mobile-SDK-Tutorials/Android-VideoStreamDecodingSample/files/786813/android-videostreamdecodingsample-debug.zip)

dji-dev, 2017-10-17:
Will go ahead and close this ticket as current version should not have this problem any more.
Please feel free to reopen if the problem still exists.
Thanks!

closed 2017-10-17 by dji-dev