DJI-Mobile-SDK-Tutorials/Android-VideoStreamDecodingSample #12: ndk-build for DJIVIDEOSTREAMDECODE: APP_ABI problems.

giannisergente, 2017-02-22:
Hi everybody, 
I've just re-compiled the source C ++, using the latest Android NDK builder, after the download of the original libffmpeg.so and after replacing the "placeholder" libffmpeg.so  located in app/jni (thanks to this post: https://github.com/DJI-Mobile-SDK-Tutorials/Android-VideoStreamDecodingSample/issues/5). 

Now I can recompile the source, but only for APP_ABI :=armeabi-v7a. This works fine with my configuration (Samsung Galaxy Grand Prime - Android 5.0.2), but the app crashes on a Nexus 7 tablet. So, i tried to recompile also the others APP_ABI, by using the following makefile:

`APP_ABI := all`

but it didn't compile correctly, with the following error message:

`C:\Users\Gianpaolo\AppData\Local\Android\Sdk\ndk-bundle\ndk-build
[arm64-v8a] SharedLibrary  : libdjivideojni.so
./obj/local/arm64-v8a/libffmpeg.so: error adding symbols: File in wrong format
clang++.exe: error: linker command failed with exit code 1 (use -v to see invocation)
make: *** [obj/local/arm64-v8a/libdjivideojni.so] Error 1
`

How can I do to compile also for the others ABI in order to try this demo on Nexus 7 tablet?

giannisergente, 2017-02-28:
No suggestions? @oliverou so, is not possible to recompile library for x86 and arm64-v8a without errors?
if I understood well, is not possible to use a modified version of your example code on devices using  x86 and arm64-v8a Application Binary Interface? please help me to fix this point...

dji-dev, 2017-10-17:
Will go ahead and close this ticket as current version should not have this problem any more.
Our app is now compatible with v7, v8a and x86.
Please feel free to reopen if the problem still exists.
Thanks!

closed 2017-10-17 by dji-dev