DJI-Mobile-SDK-Tutorials/Android-VideoStreamDecodingSample #13: showToast "disconnected"

hxl-dy, 2017-03-07:
run this sample on Huawei Tablet coonnected to remote controller of Phantom 4, the app show "disconnected"

Michael-DJI, 2017-03-17:
@hxl-dy Please provide the following infos for us to help investigate the issue:
-  Steps to reproduce the bug or crash issues(It would be great if you can provide this)
-  DJI Android SDK version you are using (Like Android SDK 3.4, etc)
-  Android device you are using (Like Samsung Galaxy Note 5, Nexus 5, etc)

hxl-dy, 2017-03-20:
if found it works when aircraft and controller both power on, but it failed when only controller power on.
Android 5.1.1 on Huawei Tablet PLE-703L

Michael-DJI, 2017-03-20:
@hxl-dy that's normal, "disconnected" is a hint message for the disconnect state of the drone

hxl-dy, 2017-03-23:
but in the code, it has three state:
1. Connected (both product and controller are running well )
2, only RC Connected (only controller connected )
3. Disconnected (neither product nor controller isn't connected )
why the second state never show?

dji-dev, 2017-10-17:
Hi there,
The second state is an app logic that uses SDK with other data to conclude.
This sample app only to show how to decode video.
I have created this gist to show an example on how to use SDK to get "RC connected" state.
https://gist.github.com/dji-dev/aabe8e204b326394aeb8930653610ce5
This is not related to this sample app so I will close this issue now.

closed 2017-10-17 by dji-dev