DJI-Mobile-SDK-Tutorials/Android-VideoStreamDecodingSample #18: 精灵3P连接成功后点“OPEN”程序直接闪退

lqsheng, 2017-04-13:
不知道为何，直接git clone下来的，只改了KEY那部分，想看看示例工程的效果，然而精灵3P连接成功后点“OPEN”程序就直接闪退，用ADB调试按下“OPEN”后的logcat如下图：
[image]
所用机型为 精灵3 Professional,手机型号是荣耀6PLUS，安卓6.0，CPU是armeabi-v7a架构，符合示例工程要求，工程的compileSdkVersion 是25

lqsheng, 2017-04-13:
进一步发现问题是工程的targetSdkVersion 不能高于22，只要高于22就会发生闪退现象，设为22及22以下就正常了，可以正常预览画面，如图：
[image]
只是点了Screen shot后，保存下来的图片没有一张是正常的，质量最好的也是下面这样的:
[image]
还有很多是下面这样的：
[image]

dji-dev, 2017-10-17:
Please use English when creating an issue

closed 2017-10-17 by dji-dev