DJI-Mobile-SDK-Tutorials/Android-VideoStreamDecodingSample #20: Demo is not running on Android Nougat

josealb, 2017-06-01:
I try to run the demo on a Nexus 6P running Android 7.1.2 and I get the following error on startup:

Video Stream Decoding Sample
Detected problems with app native libraries (please consult log for detail):
libffmpeg.so: text relocations

Can you check this out please?

Thank you!

oliverou, 2017-06-02:
Hi @josealb , which DJI Product do you use for testing? Can you provide detail logs for us to investigate? Thanks!

josealb, 2017-06-02:
Hello Oliver,

I was going to use it on a Phantom 4 Pro, but the error comes out before I even connect.
Now I am using a Nexus 5 with android 6 and it is working correctly

Thank you for your quick response

oliverou, 2017-06-02:
No problem!

Djacob502, 2017-06-16:
Just to add a bit more on this. I was getting the text relocation error and I thought the issue was libffmpeg.so and android 7. I also thought the fix was to upgrade the ffmpeg to the latest version to fix the issue. When I use the DJI sample code I also get the text relocation error. However I created a library of the video decoding sample here and dropped it into another project and the text relocation error went away. Not sure why it corrected the problem but it is nice not to have to click through the text relocation error to run the program.

dji-dev, 2017-10-17:
Will go ahead and close this ticket as current version should have a fix for this already.
Please feel free to reopen if the problem still exists.
Thanks!

closed 2017-10-17 by dji-dev