DJI-Mobile-SDK-Tutorials/Android-VideoStreamDecodingSample #22: Save the buffered data into a JPG image file will delay

jamesxienctu, 2017-10-22:
Hi, all:
When I press "Screen Shot" button, it will save buffered data into a JPG image file steadily.
The frequency is about one JPG file per second. 
Does MainActivity.screenShot() method have code to limit the freqeuncy?
Or my Android device causes this delay problem?

DJI Android SDK version is 3.5.1.
My DJI product is Phantom 3 Professional.
My Android system version is Android 6.0.
My Android device is Nexus 7.
My Android Studio version is 2.2.3.
[image]

oliverou, 2017-11-06:
Hi @jamesxienctu, we help to fix some bugs in the sample, please get the latest code and check if the issues still exist.

Djacob502, 2017-11-08:
Using the latest version, I'm getting around 30 frames per second. I'm wondering if other folks are getting the same frame rate.

jamesxienctu, 2017-12-24:
Okay, I will try. 
Thank you.