DJI-Mobile-SDK-Tutorials/Android-VideoStreamDecodingSample #23: Image Broken Problem HIGH

lillogoal, 2017-11-16:
Hello,
i tryed to implement your decoding sample.
I think I have implemented it correctly.

MY DEVICE IS: NEXUS 5X and OSMO CAMERA.

NEXUS 5x android version 8.0
and also
Samsung galaxy s7 with android 7.0

I USED THE LAST SDK 4.3.2

After decoding in FFMPEG  image results is broken or not OK.(you can see attached).

Thanks in advance
[image]

kapejod, 2017-12-08:
I can confirm this on Android 6 and 7 with different DJI products (Mavic, Osmo+).

kapejod, 2017-12-12:
If i set useSurface in MainActivity.java to false the live stream is fine.

derouck, 2017-12-12:
Problem exists on Android 8 as well, Nexus 5X, Phantom 3 Pro

Michael-DJI, 2018-05-11:
we will update this demo to MSDK 4.5.1 pls try with new version.