DJI-Mobile-SDK-Tutorials/Android-VideoStreamDecodingSample #26: My Opinion About Decoding Problem

lillogoal, 2017-12-12:
Hi,
i have been tested decoding sample with my more device mobile:
-nexus 5x
-samsung galaxy s7
-moto g5 plus
-Nvidia Shield K1

and WITH OSMO CAMERA. with this... there is a problem with decoding so Image is broken and it has noise error.

Another times i have been test with phantom 4. And DECODING IS OK.

What do you think ? I need to use decoding with my osmo mobile. Is there necessary that change my device osmo mobile with another new ?

Is Possibile that osmo mobile doesn't support perfectly DJI SDK v4 +  ? 

Thanks

derouck, 2017-12-12:
We have problems with this sample on Phantom3 Pro as well

lillogoal, 2017-12-12:
You have to use old version of SDK.
V.3.3.0 i think.

LIKE DJI GO and not like DJI GO 4

amplogik, 2018-03-23:
I have the same issue with a phantom 3 standard.   

Lillogoal,  are you saying that the glitching can be solved by downgrading the SDK we are using to a previous version?   I guess it's worth a try, but it's quite surprising.   I will try tomorrow and see

TheRedMudder, 2018-06-21:
I know this thread is old, but did you end up getting it to work?

amplogik, 2018-06-21:
I never did,  what I ended up doing was using a surfaceview holder with a listener on it and put the textureview from the demo application on it, and then trapped frames with the listener.  Direct decoding as per the sample decoding app simply doesn't appear to work for me on a Phantom 3.   I get partial images with pink banding on them or noise.   However if I just bind the stream to a texture view, I can trap frames by using a surface listener as I said.   It's not ideal as I have to have the video stream in the foreground and eating up my display,  but it's better than nothing.  I think there is some issue with the decoding sample that they haven't looked into properly.

TheRedMudder, 2018-06-21:
Thanks Kim, that’s a very helpful solution. I appreciate it.