DJI-Mobile-SDK-Tutorials/Android-VideoStreamDecodingSample #29: NativeHelper methods does not work.

saikrishnarao2, 2018-02-03:
#### Is this issue about bugs or crash issues of the Sample Code?

Yep, then please delete this template and provide the following infos for us to help investigate the issue:

-  Description of the issue.
-  Steps to reproduce the bug or crash issues(It would be great if you can provide this)
-  Crash logs (If you can find it, it would be very helpful)
-  DJI Android SDK version you are using (Like Android SDK 3.4, etc)
-  DJI Product you are using (Like Phantom 4, Mavic Pro, etc)
-  Android system version you are using (Like Android 6.0.1, Android 5.1.1, etc)
-  Android device you are using (Like Samsung Galaxy Note 5, Nexus 5, etc)
-  Android Studio version you are using (Like Android Studio 2.2, etc)
No implementation found for boolean com.dji.sdk.sample.media.NativeHelper.parse(byte[], int) (tried Java_com_dji_sdk_sample_media_NativeHelper_parse and Java_com_dji_sdk_sample_media_NativeHelper_parse___3BI)

oliverou, 2018-02-05:
Hi @saikrishnarao2 , could you help to provide the following infos for us to help investigate the issue:

-  Steps to reproduce the bug or crash issues(It would be great if you can provide this)
-  Crash logs (If you can find it, it would be very helpful)
-  DJI Android SDK version you are using (Like Android SDK 3.4, etc)
-  DJI Product you are using (Like Phantom 4, Mavic Pro, etc)
-  Android system version you are using (Like Android 6.0.1, Android 5.1.1, etc)
-  Android device you are using (Like Samsung Galaxy Note 5, Nexus 5, etc)
-  Android Studio version you are using (Like Android Studio 2.2, etc)

Thanks!

IgMart, 2018-03-14:
Hi @saikrishnarao2 , 
You need to change the name of the method with the new route of your project.
Java_new_route_project_media_NativeHelper_parse 
In every function in dji_video_jni.c and then recompile with ndk-build in order to generate new .so files.

Regards

mordka, 2018-05-08:
You also need to point the *.so libs in the gradle config.
   
```
android {
 sourceSets {
        main {
            jniLibs.srcDir 'libs'
        }
    }
}
```

jeryini, 2021-01-25:
Hello @IgMart !

I've updated function names in `dji_video_jni.c` to reflect different package name for `NativeHelper` location in my project. I'm using the following command to recompile: `/home/jernej/Android/Sdk/ndk/22.0.7026061/ndk-build` and get the following output:
```
jernej@jernej-ThinkPad-T480:~/git/drone-control-android/jni$ /home/jernej/Android/Sdk/ndk/22.0.7026061/ndk-build
/home/jernej/Android/Sdk/ndk/22.0.7026061/build/core/build-binary.mk:751: warning: overriding recipe for target '/home/jernej/git/drone-control-android/obj/local/armeabi-v7a/libffmpeg.so'
/home/jernej/Android/Sdk/ndk/22.0.7026061/build/core/build-binary.mk:751: warning: ignoring old recipe for target '/home/jernej/git/drone-control-android/obj/local/armeabi-v7a/libffmpeg.so'
/home/jernej/Android/Sdk/ndk/22.0.7026061/build/core/build-binary.mk:775: warning: overriding recipe for target '/home/jernej/git/drone-control-android/libs/armeabi-v7a/libffmpeg.so'
/home/jernej/Android/Sdk/ndk/22.0.7026061/build/core/build-binary.mk:775: warning: ignoring old recipe for target '/home/jernej/git/drone-control-android/libs/armeabi-v7a/libffmpeg.so'
/home/jernej/Android/Sdk/ndk/22.0.7026061/build/core/build-binary.mk:751: warning: overriding recipe for target '/home/jernej/git/drone-control-android/obj/local/x86/libffmpeg.so'
/home/jernej/Android/Sdk/ndk/22.0.7026061/build/core/build-binary.mk:751: warning: ignoring old recipe for target '/home/jernej/git/drone-control-android/obj/local/x86/libffmpeg.so'
/home/jernej/Android/Sdk/ndk/22.0.7026061/build/core/build-binary.mk:775: warning: overriding recipe for target '/home/jernej/git/drone-control-android/libs/x86/libffmpeg.so'
/home/jernej/Android/Sdk/ndk/22.0.7026061/build/core/build-binary.mk:775: warning: ignoring old recipe for target '/home/jernej/git/drone-control-android/libs/x86/libffmpeg.so'
/home/jernej/Android/Sdk/ndk/22.0.7026061/build/core/build-binary.mk:751: warning: overriding recipe for target '/home/jernej/git/drone-control-android/obj/local/arm64-v8a/libffmpeg.so'
/home/jernej/Android/Sdk/ndk/22.0.7026061/build/core/build-binary.mk:751: warning: ignoring old recipe for target '/home/jernej/git/drone-control-android/obj/local/arm64-v8a/libffmpeg.so'
/home/jernej/Android/Sdk/ndk/22.0.7026061/build/core/build-binary.mk:775: warning: overriding recipe for target '/home/jernej/git/drone-control-android/libs/arm64-v8a/libffmpeg.so'
/home/jernej/Android/Sdk/ndk/22.0.7026061/build/core/build-binary.mk:775: warning: ignoring old recipe for target '/home/jernej/git/drone-control-android/libs/arm64-v8a/libffmpeg.so'
[armeabi-v7a] Install        : libdjivideojni.so => libs/armeabi-v7a/libdjivideojni.so
[armeabi-v7a] Install        : libffmpeg.so => libs/armeabi-v7a/libffmpeg.so
[armeabi-v7a] Install        : libyuv2.so => libs/armeabi-v7a/libyuv2.so
[x86] Install        : libdjivideojni.so => libs/x86/libdjivideojni.so
[x86] Install        : libffmpeg.so => libs/x86/libffmpeg.so
[x86] Install        : libyuv2.so => libs/x86/libyuv2.so
[arm64-v8a] Install        : libdjivideojni.so => libs/arm64-v8a/libdjivideojni.so
[arm64-v8a] Install        : libffmpeg.so => libs/arm64-v8a/libffmpeg.so
[arm64-v8a] Install        : libyuv2.so => libs/arm64-v8a/libyuv2.so
```

I can see that `.so` files have been built successfully, but when I run the example, the main surface video is not working anymore. Should I use some additional options when recompiling?

DJI-William, 2022-03-01:
Is problem solved?