DJI-Mobile-SDK-Tutorials/Android-VideoStreamDecodingSample #30: Error on startup

Rishat7c, 2018-03-06:
When the application starts, it displays the Register SDK Succes dialog box and closes the application.

Fears on these three lines in the ConnectionActivity class:

```
1. UserAccountManager.getInstance (). LogIntoDJIUserAccount (...)
2. It collapses into the class itself
3. loginDJIUserAccount ()
```

[image]

Michael-DJI, 2018-03-07:
better to turn off shrinkResources like following:
shrinkResources false

Rishat7c, 2018-03-07:
There is no parameter in the gradle.

```
buildscript {
    repositories {
        jcenter()
        google()
    }
    dependencies {
        classpath 'com.android.tools.build:gradle:3.0.1'
    }
}
apply plugin: 'com.android.application'

allprojects {
    repositories {
        jcenter()
        google()
    }
}
android {
    compileSdkVersion 24
    buildToolsVersion '26.0.2'

    defaultConfig {
        applicationId "com.dji.videostreamdecodingsample"
        minSdkVersion 16
        targetSdkVersion 24
        multiDexEnabled true
    }
    buildTypes {
        release {
            minifyEnabled false
            proguardFiles getDefaultProguardFile('proguard-android.txt'), 'proguard-rules.pro'
        }
    }
    sourceSets {
        main {
            jniLibs.srcDir 'libs'
        }
    }

    packagingOptions {
        doNotStrip "*/*/libdjivideo.so"
        doNotStrip "*/*/libSDKRelativeJNI.so"
        doNotStrip "*/*/libFlyForbid.so"
        doNotStrip "*/*/libduml_vision_bokeh.so"
        doNotStrip "*/*/libyuv2.so"
        doNotStrip "*/*/libGroudStation.so"
        doNotStrip "*/*/libFRCorkscrew.so"
        doNotStrip "*/*/libUpgradeVerify.so"
        doNotStrip "*/*/libFR.so"
    }
}

dependencies {
    compile fileTree(include: ['*.jar'], dir: 'libs')
    compile 'com.android.support:multidex:1.0.1'
    compile 'com.android.support:appcompat-v7:24.2.0'
    compile 'com.dji:dji-sdk:4.4.1'
    provided 'com.dji:dji-sdk-provided:4.4.1'
}
```

Rishat7c, 2018-03-09:
The problem is still relevant, are there solutions?

Michael-DJI, 2018-03-15:
@Rishat7c which phone pls?

Rishat7c, 2018-03-15:
@Michael-DJI corrected the problem by commenting out one line in ConnectionActivity
this :
`loginDJIUserAccount();`

closed 2018-03-15 by Rishat7c

Michael-DJI, 2018-03-19:
@Rishat7c yes could provide the device you were using? so that we could fix some potential issue.

Rishat7c, 2018-03-19:
Xiaomi Redmi 4X

Michael-DJI, 2018-05-11:
we will update the sample with MSDK v4.5.1 in nearly days, pls try with it then.