DJI-Mobile-SDK-Tutorials/Android-VideoStreamDecodingSample #34: DJICodecManager.YuvDataCallback makes the preview stuck

mizhoux, 2018-05-31:
#### Bug

As long as I click the button "YUV Screen Shot", the preview will be stuck and then the screen shots will all be the same. 

My devices' information:
-  DJI Android SDK version: 4.5.1
-  DJI Product:  Phantom 4 Pro
-  Android system version: 7.1.1
-  Android Studio version: 3.1.1

slamgo, 2018-08-27:
我也遇到这个问题，请问你是怎么解决的

mizhoux, 2019-03-20:
即使在目前最新的 SDK 4.9 上，这个问题也依然没有解决。我不知道为什么大疆的开发者会对一个这么明显的大 bug 在如此长的时间之内置若罔闻，至少 YUVDataCallable 从未在 Phantom 4 Pro 正常工作过，我对大疆的 SDK 开发者感到极度的失望。

Michael-DJI, 2019-03-21:
@mizhoux 您好，我们现在的正式反馈渠道是dev@dji.com, 由于人力问题，这里的信息有时候不会及时的被处理，如果后续有问题请尽量通过正式渠道反馈，另外我们也提供一些高优先级的技术支持服务，但是需要支付一定费用，您也可以考虑使用，对于这个问题，我们现在会反馈到开发，开发可能需要一段时间去处理，再次感谢您的支持！

neilyoung, 2019-05-18:
@mizhoux @Michael-DJI Would it be possible to repeat your comments in English? I'm having the same issue.