DJI-Mobile-SDK-Tutorials/Android-VideoStreamDecodingSample #37: Streaming in RTMP

yulvita, 2018-06-12:
Hey,

Would that be possible for you guys to drop any information or sample about a custom RTMP implementation for this app ?

Want to send a RTMP stream in a push configuration to the server while keeping the video preview displayed onto the Android Surface.

It's basically the same thing (think it's called "Custom RTMP") as in the DJI Go app (but as I can't find the open-sourced version of it, I'm looking for some help).

Thanks

zhanglibo007, 2018-12-19:
Have u sent video strem to server by rtmp ？

StrongZuo, 2019-01-25:
Can you achieve it?
How can I achieve it?

StrongZuo, 2019-01-26:
When will the release of 4.9?

Michael-DJI, 2019-01-29:
4.9 has been released now, pls check the release notes:
https://developer.dji.com/mobile-sdk/

Michael-DJI, 2019-03-21:
here is the sample code of how to stream the video to a RTMP server:
https://github.com/dji-sdk/Mobile-SDK-Android/blob/master/Sample%20Code/app/src/main/java/com/dji/sdk/sample/demo/camera/LiveStreamView.java