DJI-Mobile-SDK-Tutorials/Android-VideoStreamDecodingSample #41: Demo custom decoder

YangShengLiQQ, 2018-08-23:
I ran the sample code and clicked demo custom decoder button. I found if I connected the remote control through the wifi, It worked well. However, If I connected the remote control through the USB line, and clicked demo custom decoder button,  the screen was black. It didn't receive the video data because I didn't find Log.d(TAG, "camera recv video data size: " + size) in logcat.  If I tried other buttons like Demo SurfaceView and Demo TextureView and connected the remote control through the USB line, They worked well. I also tried other cellphone to test it, and the result was the same.

Michael-DJI, 2018-08-23:
which drone?

YangShengLiQQ, 2018-08-23:
spark

YangShengLiQQ, 2018-08-30:
I ran the lastest version sample code, and it worked well. My project use older SDK(4.5.1). If I use the latest SDK, the problem will be solved? Or just change some code?

Michael-DJI, 2018-08-31:
in 4.7, decoding sample use data get from provideTranscodedVideoFeed() to feed into the custom decoder, could u explain what's your feature?

oscarmore2, 2018-12-05:
But TranscodedVideoFeed() was never called when in USB connection with Maivc Pro and Maivc Zoom. My sdk version is 4.8.1.

neilyoung, 2019-05-16:
Here initCodec() immediately returns, since width and height are 0.

If I fake a w/h it is stuck a bit later 

W/ACodec: [OMX.qcom.video.decoder.avc] does not support color format 2130708361

What a mess...