DJI-Mobile-SDK-Tutorials/Android-VideoStreamDecodingSample #44: Input Video Buffer

abhijitnathwani, 2018-11-19:
Hi,

I was able to successfully execute this project, however, I'm left with certain questions.
1. What is the input VideoBuffer format so that it decodes to YUV data?
2. What are the changes required to be able to stream to RTMP/ RTSP?
3. If the input videoBuffer from the drone is h.264 encoded, why can't we play it when we dump those frames to a file?

Please shed some light here. I'm stuck.

Regards,
Abhijit Nathwani

Michael-DJI, 2019-03-21:
@abhijitnathwani hey, for most drones, like mavic the data of live stream is standard h264 (annex b) format, in 4.9 we have some interfaces which could stream the video to a RTMP server, pls check the sample code here:

IgMart, 2021-07-09:
Hi @Michael-DJI ;
Thank you for your answer. 
Could we go deeper into question number 3? @abhijitnathwani  
> If the input videoBuffer from the drone is h.264 encoded, why can't we play it when we dump those frames to a file?

I am using the following code to save the bytestream to "video.h264". Later, I try to reproduce that video file with FFMPEG. However, FFMPEG does not recognise the bytestream (error: "non-existing PPS 0 referenced" and "decode_slice_header error").
```
// The callback for receiving the raw H264 video data for camera live view
        mReceivedVideoDataListener = new VideoFeeder.VideoDataListener() {
            @Override
            public void onReceive(byte[] videoBuffer, int size) {
                File file = new File(PATH+"/video.h264");
                try {
                    if (!file.exists()) {
                        file.createNewFile();
                    }
                    FileOutputStream fos = new FileOutputStream(file, true);
                    fos.write(videoBuffer);
                    fos.close();
                } catch (Exception e) {
                    Log.e(TAG, e.getMessage());
                }
```
Please, I would really appreciate your support with this issue. 
Kind Regards

neilyoung, 2021-07-09:
The buffer is not "just video". It is H.264 encoded stuff, means it contains SPS, PPS and a lot of other meta-data alongside with the video.

paglySalva, 2021-07-09:
> The buffer is not "just video". It is H.264 encoded stuff, means it contains SPS, PPS and a lot of other meta-data alongside with the video.

Thank you @neilyoung, 
I've tried the following:

`ffmpeg -framerate 29.97 -i video.264 -c copy video.mp4` 

But still not working.
Any idea?

Thanks in advance.

neilyoung, 2021-07-09:
Yes. The idea is to go and run an H.264 decoder over the bitstream in order to separate the video from the jabber