DJI-Mobile-SDK-Tutorials/Android-VideoStreamDecodingSample #45: Unable to login and register product after I use usb connect to pad. Can't get videofeed via USB

oscarmore2, 2018-11-30:
Hi I download the project and it was build successfully. however I run the project at the phone. the aircraft can't be discover, and it said "Register sdk fails, check network is available". But I had connect to my WIfi.

I am in china, I log in my user name in Dji go. but it does't work in the this version 4.8 project.

I also try to use an exist Logged in project to get videofeeder via USB, but the get Feeder callback doesn't call at all.

Wait for your reply.

#### Is this issue about bugs or crash issues of the Sample Code?

Yep, then please delete this template and provide the following infos for us to help investigate the issue:

-  Description of the issue.
-  Steps to reproduce the bug or crash issues(It would be great if you can provide this)
-  Crash logs (If you can find it, it would be very helpful)
-  DJI Android SDK version you are using (Like Android SDK 3.4, etc)
-  DJI Product you are using (Like Phantom 4, Mavic Pro, etc)
-  Android system version you are using (Like Android 6.0.1, Android 5.1.1, etc)
-  Android device you are using (Like Samsung Galaxy Note 5, Nexus 5, etc)
-  Android Studio version you are using (Like Android Studio 2.2, etc)

oscarmore2, 2018-12-01:
2018-12-01 15:05:54.520 17582-17598/? W/LocalConnector: Cannot connect the wmserver-data-reciever
2018-12-01 15:05:55.329 29671-29677/com.flouav.flopilot I/zygote64: Do full code cache collection, code=502KB, data=312KB
2018-12-01 15:05:55.336 29671-29677/com.flouav.flopilot I/zygote64: After code cache collection, code=453KB, data=242KB
2018-12-01 15:05:55.361 1683-24092/? I/WifiService: getDhcpInfo uid=10062
2018-12-01 15:05:55.362 1683-24092/? I/WifiService: getConnectionInfo uid=10062
2018-12-01 15:05:55.363 1683-24092/? I/WifiService: getWifiEnabledState uid=10062
2018-12-01 15:05:55.365 1683-24092/? I/WifiService: startScan uid=10062
2018-12-01 15:05:55.368 14464-14491/? I/NetworkLocationManager: start get current Location
2018-12-01 15:05:55.369 14464-14464/? I/MetokLocationProvider: report position from 2
2018-12-01 15:05:55.373 1683-3043/? D/NpProxy: handleMessage what - 4
2018-12-01 15:05:55.374 1683-3043/? D/NpProxy: Location: Location[network 31.198794,121.432881 hAcc=40 et=+3d3h31m37s680ms vAcc=??? sAcc=??? bAcc=??? {Bundle[{networkLocationType=wifi, source=2}]}]
2018-12-01 15:05:55.374 1683-3043/? V/LocSvc_HIDL_OsNpGlue_jni: [onJavaLocationChanged][177] [HC] =>> [HS]
2018-12-01 15:05:55.381 753-753/? V/LocSvc_HIDL_IzatOsNpGlue: [locationChanged][47] [HS] <<<<= [HC]
2018-12-01 15:05:55.521 17582-17598/? W/LocalConnector: Cannot connect the wmserver-data-reciever

Michael-DJI, 2019-03-21:
@oscarmore2 hey, for register issue, the http connection of register will have some problem occasionally when using 4G of china mobile, please try to connect to a wifi network and retry.

oscarmore2, 2019-09-04:
[Update]: I Pulled the lastest version of Video decoding simple, which includeded dji sdk 4.10. Then build a app to my huwawei m6 pad, and connect it to marvic pro, I found the listener from provideTranscodedVideoFeed() feeder is never invoked at all, and the demo custom decoder surface is totally black. But when I connect to marvic 2, it's  I could easily fetch the raw video data for pushing or caching.
[https://cdn.flouav.com/screenshot/WechatIMG1350.jpeg](https://cdn.flouav.com/screenshot/WechatIMG1350.jpeg)