DJI-Mobile-SDK-Tutorials/Android-VideoStreamDecodingSample #46: Unable to get video data from Mavic 2 via DJICodecManager in SDK 4.8

oscarmore2, 2018-12-05:
I encounter another problem from SDK 4.8.
With failure of getting the data from VideoFeeder. I start tried to use CodecManager and implement the interface onYuvDataReceived and register to DJICodecManager instance.

firstly, it works very well in Mavic Pro via .USE can get the data from onYuvDataReceived, and render the Yuv data to texture.

But when I switch the Usb to the Mavic 2, the onYuvDataReceived interface was never call. Any idea which direction I should look on?

Michael-DJI, 2019-03-21:
@oscarmore2 hey, what's the feature you were developing if you want to steaming the video to a RTMP server please use the interfaces in LiveStreamManager class here is the sample code:

oscarmore2, 2019-09-02:
what about 4.8 version