DJI-Mobile-SDK-Tutorials/Android-VideoStreamDecodingSample #51: Could not find com.dji:dji-sdk-provided:4.9.

hxl-dy, 2019-04-18:
#### Is this issue about bugs or crash issues of the Sample Code?

Yep, then please delete this template and provide the following infos for us to help investigate the issue:

-  Description of the issue.
Could not find com.dji:dji-sdk-provided:4.9.

-  Steps to reproduce the bug or crash issues(It would be great if you can provide this)
1. git clone
2. open with android studio
3. make project

-  Crash logs (If you can find it, it would be very helpful)
Could not find com.dji:dji-sdk-provided:4.9.
Searched in the following locations:
    https://dl.google.com/dl/android/maven2/com/dji/dji-sdk-provided/4.9/dji-sdk-provided-4.9.pom
    https://dl.google.com/dl/android/maven2/com/dji/dji-sdk-provided/4.9/dji-sdk-provided-4.9.jar
Required by:
    project :app

-  DJI Android SDK version you are using (Like Android SDK 3.4, etc)
Android SDK 4.9

-  DJI Product you are using (Like Phantom 4, Mavic Pro, etc)
-  Android system version you are using (Like Android 6.0.1, Android 5.1.1, etc)
-  Android device you are using (Like Samsung Galaxy Note 5, Nexus 5, etc)
-  Android Studio version you are using (Like Android Studio 2.2, etc)
Android Studio 3.4

ustcwzy1986, 2019-04-26:
I met same problem.

ERROR: Failed to resolve: com.dji:dji-sdk:4.9
ERROR: Failed to resolve: com.dji:dji-sdk-provided:4.9

Michael-DJI, 2019-04-30:
@hxl-dy could you please show us the full error message? btw what's the version of you Android Studio? could you please upgrade it to the latest stable version then  retry?

hxl-dy, 2019-04-30:
> @hxl-dy could you please show us the full error message? btw what's the version of you Android Studio? could you please upgrade it to the latest stable version then retry?

All message I got are shown above in bold.

Michael-DJI, 2019-04-30:
allprojects {
    repositories {
        google()
        jcenter()
        
    }
}
add jcenter()

closed 2019-04-30 by Michael-DJI