DJI-Mobile-SDK-Tutorials/Android-VideoStreamDecodingSample #54: with msdk-v4.10 runtime error

hxl-dy, 2019-05-18:
#### Is this issue about bugs or crash issues of the Sample Code?

Yep, then please delete this template and provide the following infos for us to help investigate the issue:

-  Description of the issue.
upgrade msdk to v4.10, build and run with error

-  Steps to reproduce the bug or crash issues(It would be great if you can provide this)
https://github.com/D-Y-Innovations/Android-VideoStreamDecodingSample/commit/21d09cf631873a0aac41a07db63c6a5067d528a5

-  Crash logs (If you can find it, it would be very helpful)
```
2019-05-18 14:15:05.358 25447-25455/? E/art: Failed sending reply to debugger: Broken pipe
2019-05-18 14:15:07.260 25447-25447/com.dji.videostreamdecodingsample E/sdk: install begin
2019-05-18 14:15:07.263 25447-25447/com.dji.videostreamdecodingsample E/sdk: v19 install
2019-05-18 14:15:07.345 25447-25447/com.dji.videostreamdecodingsample E/sdk: install begin
2019-05-18 14:15:07.346 25447-25447/com.dji.videostreamdecodingsample E/sdk: v19 install
2019-05-18 14:15:07.414 25447-25447/com.dji.videostreamdecodingsample E/linker: "/data/app/com.dji.videostreamdecodingsample-2/lib/arm64/libsfjni.so" .dynamic section has invalid link(0) sh_type: 0 (expected SHT_STRTAB)
2019-05-18 14:15:07.416 25447-25447/com.dji.videostreamdecodingsample E/sfjni: Couldn't load lib
2019-05-18 14:15:07.416 25447-25447/com.dji.videostreamdecodingsample E/art: No implementation found for void com.dji.megatronking.stringfog.StringUtilsJNI.test() (tried Java_com_dji_megatronking_stringfog_StringUtilsJNI_test and Java_com_dji_megatronking_stringfog_StringUtilsJNI_test__)
2019-05-18 14:15:07.418 25447-25447/com.dji.videostreamdecodingsample E/AndroidRuntime: FATAL EXCEPTION: main
    Process: com.dji.videostreamdecodingsample, PID: 25447
    java.lang.UnsatisfiedLinkError: No implementation found for void com.dji.megatronking.stringfog.StringUtilsJNI.test() (tried Java_com_dji_megatronking_stringfog_StringUtilsJNI_test and Java_com_dji_megatronking_stringfog_StringUtilsJNI_test__)
        at com.dji.megatronking.stringfog.StringUtilsJNI.test(Native Method)
        at com.dji.megatronking.stringfog.a.c.b(Unknown Source)
        at dji.midware.e.b(Unknown Source)
        at dji.keysdk.DJIKey.<clinit>(Unknown Source)
        at com.dji.videostreamdecodingsample.ConnectionActivity.<init>(ConnectionActivity.java:72)
        at java.lang.Class.newInstance(Native Method)
        at android.app.Instrumentation.newActivity(Instrumentation.java:1078)
        at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2558)
        at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:2727)
        at android.app.ActivityThread.-wrap12(ActivityThread.java)
        at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1478)
        at android.os.Handler.dispatchMessage(Handler.java:102)
        at android.os.Looper.loop(Looper.java:154)
        at android.app.ActivityThread.main(ActivityThread.java:6121)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:912)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:802)
```

-  DJI Android SDK version you are using (Like Android SDK 3.4, etc)
Android SDK 4.9

-  DJI Product you are using (Like Phantom 4, Mavic Pro, etc)
P4P

-  Android system version you are using (Like Android 6.0.1, Android 5.1.1, etc)
Android 7.1.2

-  Android device you are using (Like Samsung Galaxy Note 5, Nexus 5, etc)
rk-3399-mid

-  Android Studio version you are using (Like Android Studio 2.2, etc)
Android Studio 3.4

samoilovich, 2019-05-30:
I have the same issue when just ask dji sdk about something.
As an example DJISDKManager.getInstance().product
If you need more details, please let me know I will create demo app with this crash.

Ray-DJI, 2019-05-30:
Hi, @hxl-dy  @samoilovich . DJI SDK with 4.10 have updated the secure strategies. Please add following lines in your project build.gradle
packagingOptions {

    doNotStrip "*/*/libdjivideo.so"
    doNotStrip "*/*/libSDKRelativeJNI.so"
    doNotStrip "*/*/libFlyForbid.so"
    doNotStrip "*/*/libduml_vision_bokeh.so"
    doNotStrip "*/*/libyuv2.so"
    doNotStrip "*/*/libGroudStation.so"
    doNotStrip "*/*/libFRCorkscrew.so"
    doNotStrip "*/*/libUpgradeVerify.so"
    doNotStrip "*/*/libFR.so"

    // Add the new so
    doNotStrip "*/*/libDJIFlySafeCore.so"
    doNotStrip "*/*/libdjifs_jni.so"
    doNotStrip "*/*/libsfjni.so"
    exclude 'META-INF/rxjava.properties'
}

Thanks for the feedback, we will update the tutorial with the SDK 4.10 as soon as possible.

PadreCocoRocha, 2019-06-01:
Same problem with me, adding these lines does not solve it.
Looks like it happens on call to DJISDKManager.getInstance().getSDKVersion()...

StackTrace:

2019-06-01 20:38:44.748 E/sfjni: Couldn't load lib
2019-06-01 20:38:44.748 E/zygote: No implementation found for void com.dji.megatronking.stringfog.StringUtilsJNI.test() (tried Java_com_dji_megatronking_stringfog_StringUtilsJNI_test and Java_com_dji_megatronking_stringfog_StringUtilsJNI_test__)
2019-06-01 20:38:44.748 D/AndroidRuntime: Shutting down VM
2019-06-01 20:38:44.753 E/AndroidRuntime: FATAL EXCEPTION: main
    Process: com.panorama_id.codecapture, PID: 20026
    java.lang.UnsatisfiedLinkError: No implementation found for void com.dji.megatronking.stringfog.StringUtilsJNI.test() (tried Java_com_dji_megatronking_stringfog_StringUtilsJNI_test and Java_com_dji_megatronking_stringfog_StringUtilsJNI_test__)
        at com.dji.megatronking.stringfog.StringUtilsJNI.test(Native Method)
        at com.dji.megatronking.stringfog.a.c.b(Unknown Source:3)
        at dji.midware.e.b(Unknown Source:2)
        at dji.sdk.sdkmanager.DJISDKManager.<clinit>(Unknown Source:2)
        at dji.sdk.sdkmanager.DJISDKManager.getInstance(Unknown Source:0)

Michael-DJI, 2019-06-03:
@PadreCocoRocha thanks for the feedback, could you please provide a public github repo? with that our developer could clone it and reproduce your issue.

chano1025, 2019-06-04:
clean project -> rebuild project

PadreCocoRocha, 2019-06-04:
> clean project -> rebuild project

That solved for me, thank you

accuware, 2019-06-04:
Out of curiosity: Where is that 4.10 SDK available? All the sample apps are still using 4.9.

Ray-DJI, 2019-06-05:
@accuware Our 4.10 SDK already have published for almost one month, and the `VideoStreamDecodingSample` have updated with the latest 4.10 SDK.

accuware, 2019-06-05:
@Ray-DJI  I see. This is surprising, since I cloned the sample a week ago. I see you have updated 5 days ago...Since you also have updated the YUV part I will check it out. Thanks

samoilovich, 2019-06-10:
@Ray-DJI 
I faced one more issue. Your previous solution works! But when I tried to test release version with turned on proguard I get the crash:
`    Process: com.measure.groundcontrol, PID: 22027
    java.lang.NoClassDefFoundError: Failed resolution of: Landroid/arch/lifecycle/LifecycleObserver;
        at dji.sdk.sdkmanager.DJISDKManager.getInstance`

Could you help with that?

samoilovich, 2019-06-10:
@Ray-DJI 
I have added keep annotation to proguard file for android architecture component from this link
https://gist.github.com/brownsoo/9b11a823360c9cf184263df3e669375c
And it helped but I think that this moment should be fixed.
Please let me know if you need any info.

Ray-DJI, 2019-06-10:
Hi, @samoilovich You are right, the architecture component should keep from obfuscating,  because our SDK AAR Package depends on the architecture component aar.
Each time when we release a new version, we will give the proguard file to tell the developer which class or method should be kept.

samoilovich, 2019-06-10:
Any dji sample app? 
hmmm. https://docker.pkg.github.com/DJI-Mobile-SDK-Tutorials/Android-VideoStreamDecodingSample
I tried to find here the proguard file. But could :(
Could you send a link?

Ray-DJI, 2019-06-11:
@samoilovich The link is below
https://github.com/dji-sdk/Mobile-SDK-Android/blob/master/Sample%20Code/app/proguard-rules.pro