DJI-Mobile-SDK-Tutorials/Android-VideoStreamDecodingSample #55: Q: Would you mind to elaborate a little change between 4.9 and 4.10?

neilyoung, 2019-06-05:
I'm talking about this routing in the sample:

First there is a bug: To test yuvFrame.length to be >= with*height is necessary, but not sufficient. Actually .length has to be tested to be exactly 1.5 * widht * height.

Then please confirm: Am I seeing this correctly 

- You are treating the incoming YUV buffer to be NV21 (https://wiki.videolan.org/YUV): Y followed by interleaved U and V, starting with V.

- Then you are making it NV12 (Y then U and V interleaved, starting with U)

- just to provide it to a function, which seams to expect NV21.

??

```
private void newSaveYuvDataToJPEG(byte[] yuvFrame, int width, int height){
        if (yuvFrame.length < width * height) {
            //DJILog.d(TAG, "yuvFrame size is too small " + yuvFrame.length);
            return;
        }
        int length = width * height;

        byte[] u = new byte[width * height / 4];
        byte[] v = new byte[width * height / 4];
        for (int i = 0; i < u.length; i++) {
            v[i] = yuvFrame[length + 2 * i];
            u[i] = yuvFrame[length + 2 * i + 1];
        }
        for (int i = 0; i < u.length; i++) {
            yuvFrame[length + 2 * i] = u[i];
            yuvFrame[length + 2 * i + 1] = v[i];
        }
        screenShot(yuvFrame,Environment.getExternalStorageDirectory() + "/DJI_ScreenShot", width, height);
    }

```

later

```
YuvImage yuvImage = new YuvImage(buf,
                ImageFormat.NV21,
                width,
                height,
                null);

```

Could you please elaborate? I don't get that back and forth and according to my experience it is not always guaranteed to have NV21. My Mavic Pro delivers I420: Y followed by U followed by V non-interleaved.

I'm having a ticket open at your support regarding this (24417). No answer yet. Would you mind to explain?

neilyoung, 2019-06-05:
And what is the mystery behind this?

```
                    if (Build.VERSION.SDK_INT <= 23){
                        oldSaveYuvDataToJPEG(bytes, width, height);
                    }else {
                        newSaveYuvDataToJPEG(bytes, width, height);
                    }

```

What has the SDK version to do with this?

neilyoung, 2019-06-05:
Oh, WTF.. .my post is scrambled. Here again:

I'm talking about this routing in the sample:

First there is a bug: To test yuvFrame.length to be >= with*height is necessary, but not sufficient. Actually .length has to be tested to be exactly 1.5 * widht * height.

Then please confirm: Am I seeing this correctly 

- You are treating the incoming YUV buffer to be NV21 (https://wiki.videolan.org/YUV): Y followed by interleaved U and V, starting with V.

- Then you are making it NV12 (Y then U and V interleaved, starting with U)

- just to provide it to a function, which seams to expect NV21.

??

 private void newSaveYuvDataToJPEG(byte[] yuvFrame, int width, int height){
        if (yuvFrame.length < width * height) {
            //DJILog.d(TAG, "yuvFrame size is too small " + yuvFrame.length);
            return;
        }
        int length = width * height;

        byte[] u = new byte[width * height / 4];
        byte[] v = new byte[width * height / 4];
        for (int i = 0; i < u.length; i++) {
            v[i] = yuvFrame[length + 2 * i];
            u[i] = yuvFrame[length + 2 * i + 1];
        }
        for (int i = 0; i < u.length; i++) {
            yuvFrame[length + 2 * i] = u[i];
            yuvFrame[length + 2 * i + 1] = v[i];
        }
        screenShot(yuvFrame,Environment.getExternalStorageDirectory() + "/DJI_ScreenShot", width, height);
    }

later

YuvImage yuvImage = new YuvImage(buf,
                ImageFormat.NV21,
                width,
                height,
                null);

I'm having a ticket open at your support regarding this (24417). No answer yet. Would you mind to explain?

Michael-DJI, 2019-06-05:
Hey @neilyoung, thanks for the feedback! we have received your ticket (24417), it will be delivered to our developer soon, if the developer give the feedback, we will reply your email ASAP, thanks! BTW we might need you to provide some info regarding this issue, please check the email later, thanks again!

dji-dev, 2019-06-05:
> Public comment from Luce Luo in <a href='https://djisdksupport.zendesk.com/agent/tickets/24678'>Zendesk ticket #24678</a>:<br><br>Dear Customer,<br><br>Thank you for contacting DJI.<br>To collect questions in detail, please fill in the form below, including your context, mobile device and android system info, and we will handle this ticket as soon as we receive it.<br>https://formcrafts.com/a/dji-developer-feedback-en<br><br>Thanks,<br><br>Luce Luo<br>DJI Dev Team

neilyoung, 2019-06-05:
@dji-dev It is not an issue, it is just a question. Maybe all the problems are gone if you answer me to 24417 :)

neilyoung, 2019-06-05:
https://github.com/accuware/djistreamerlib

Maybe this is of interest in this case