DJI-Mobile-SDK-Tutorials/Android-VideoStreamDecodingSample #59: Does not work with 4.11

neilyoung, 2019-09-24:
as subject says.

Didn't find class "androidx.lifecycle.LifecycleObserver"

neilyoung, 2019-09-24:
OK, let's be a bit more specific.

- Cloned the repo into a new directory. Compiled with 4.10. Works.
- Changed `build.gradle`

```
implementation('com.dji:dji-sdk:4.11', {
        //if your app needn't support Mavic 2 Pro and Mavic 2 Zoom, pls exclude the library-anti-distortion module
        //like following
        //exclude module: 'library-anti-distortion'

        //Uncomment the following line if your APP does not need network RTK.
        //For Phantom 4 RTK and Matrice 210 RTK V2. Network RTK can be used as a virtual reference station to provide
        //cm level accurate position.
        //exclude module: 'library-networkrtk-helper'
    })
    compileOnly 'com.dji:dji-sdk-provided:4.11'

```

-  DJIVideoStreamDecoder.java: Added dummy for first parameter, don't know where to get from, don't need either.

`yuvDataListener.onYuvDataReceived(null, yuvDataBuf, bufferInfo.size - bufferInfo.offset,  width, height);
`
- ConnectionActivity.java: Added new delegate for `DJISDKManager.SDKManagerCallback()`
```

 @Override
 public void onDatabaseDownloadProgress(long l, long l1) {
  // No idea what to do here
 }
```

- MainActivity.java: Added parameter `format`:
```
   @Override
    public void onYuvDataReceived(MediaFormat format, final ByteBuffer yuvFrame, int dataSize, final int width, final int height) {
 }

```

- Compile, install. App crashes very early in  

```
DJISDKManager.getInstance().registerApp(ConnectionActivity.this.getApplicationContext(), new DJISDKManager.SDKManagerCallback() {
                   }

```

with 

```
2019-09-24 21:06:54.432 31128-31160/? I/zygote64: Rejecting re-init on previously-failed class java.lang.Class<dji.sdk.sdkmanager.DJISDKManager$DJILifecycleListener>: java.lang.NoClassDefFoundError: Failed resolution of: Landroidx/lifecycle/LifecycleObserver;
2019-09-24 21:06:54.432 31128-31160/? I/zygote64:     at dji.sdk.sdkmanager.DJISDKManager dji.sdk.sdkmanager.DJISDKManager.getInstance() (:-1)
2019-09-24 21:06:54.432 31128-31160/? I/zygote64:     at void com.dji.videostreamdecodingsample.ConnectionActivity$2.run() (ConnectionActivity.java:106)
2019-09-24 21:06:54.432 31128-31160/? I/zygote64:     at void android.os.AsyncTask$SerialExecutor$1.run() (AsyncTask.java:245)
2019-09-24 21:06:54.432 31128-31160/? I/zygote64:     at void java.util.concurrent.ThreadPoolExecutor.runWorker(java.util.concurrent.ThreadPoolExecutor$Worker) (ThreadPoolExecutor.java:1162)
2019-09-24 21:06:54.432 31128-31160/? I/zygote64:     at void java.util.concurrent.ThreadPoolExecutor$Worker.run() (ThreadPoolExecutor.java:636)
2019-09-24 21:06:54.432 31128-31160/? I/zygote64:     at void java.lang.Thread.run() (Thread.java:764)
2019-09-24 21:06:54.432 31128-31160/? I/zygote64: Caused by: java.lang.ClassNotFoundException: Didn't find class "androidx.lifecycle.LifecycleObserver" on path: DexPathList[[zip file "/data/app/com.dji.videostreamdecodingsample-ruAaKdDxqK_EK8AS5WU-AQ==/base.apk", zip file "/data/app/com.dji.videostreamdecodingsample-ruAaKdDxqK_EK8AS5WU-AQ==/split_lib_dependencies_apk.apk", zip file "/data/app/com.dji.videostreamdecodingsample-ruAaKdDxqK_EK8AS5WU-AQ==/split_lib_resources_apk.apk", zip file "/data/app/com.dji.videostreamdecodingsample-ruAaKdDxqK_EK8AS5WU-AQ==/split_lib_slice_0_apk.apk", zip file "/data/app/com.dji.videostreamdecodingsample-ruAaKdDxqK_EK8AS5WU-AQ==/split_lib_slice_1_apk.apk", zip file "/data/app/com.dji.videostreamdecodingsample-ruAaKdDxqK_EK8AS5WU-AQ==/split_lib_slice_2_apk.apk", zip file "/data/app/com.dji.videostreamdecodingsample-ruAaKdDxqK_EK8AS5WU-AQ==/split_lib_slice_3_apk.apk", zip file "/data/app/com.dji.videostreamdecodingsample-ruAaKdDxqK_EK8AS5WU-AQ==/split_lib_slice_4_apk.apk", zip file "/data/app/com.dji.videostreamdecodingsample
2019-09-24 21:06:54.432 31128-31160/? I/zygote64:     at java.lang.Class dalvik.system.BaseDexClassLoader.findClass(java.lang.String) (BaseDexClassLoader.java:125)
2019-09-24 21:06:54.432 31128-31160/? I/zygote64:     at java.lang.Class java.lang.ClassLoader.loadClass(java.lang.String, boolean) (ClassLoader.java:379)
2019-09-24 21:06:54.432 31128-31160/? I/zygote64:     at java.lang.Class java.lang.ClassLoader.loadClass(java.lang.String) (ClassLoader.java:312)
2019-09-24 21:06:54.432 31128-31160/? I/zygote64:     at dji.sdk.sdkmanager.DJISDKManager dji.sdk.sdkmanager.DJISDKManager.getInstance() (:-1)
2019-09-24 21:06:54.432 31128-31160/? I/zygote64:     at void com.dji.videostreamdecodingsample.ConnectionActivity$2.run() (ConnectionActivity.java:106)
2019-09-24 21:06:54.432 31128-31160/? I/zygote64:     at void android.os.AsyncTask$SerialExecutor$1.run() (AsyncTask.java:245)
2019-09-24 21:06:54.432 31128-31160/? I/zygote64:     at void java.util.concurrent.ThreadPoolExecutor.runWorker(java.util.concurrent.ThreadPoolExecutor$Worker) (ThreadPoolExecutor.java:1162)
2019-09-24 21:06:54.432 31128-31160/? I/zygote64:     at void java.util.concurrent.ThreadPoolExecutor$Worker.run() (ThreadPoolExecutor.java:636)
2019-09-24 21:06:54.432 31128-31160/? I/zygote64:     at void java.lang.Thread.run() (Thread.java:764)
2019-09-24 21:06:54.447 31128-31160/? E/AndroidRuntime: FATAL EXCEPTION: DJIExecutor Urgent #1
 
```

No idea, how to solve this:`Didn't find class "androidx.lifecycle.LifecycleObserver`

Your QA may thank me later.

dji-dev, 2019-09-25:
Public comment from Luce Luo in <a href='https://djisdksupport.zendesk.com/agent/tickets/28278'>Zendesk ticket #28278</a>:<br><br>Dear Neil,

Thank you for contacting DJI.
Please refer to this sample to add the androidx supported into your project.
https://github.com/dji-sdk/Mobile-SDK-Android/blob/master/Sample%20Code/app/build.gradle

implementation 'androidx.appcompat:appcompat:1.0.0'
    implementation 'androidx.core:core:1.0.0'
    implementation 'androidx.constraintlayout:constraintlayout:1.1.3'
    implementation 'androidx.recyclerview:recyclerview:1.0.0'
    implementation 'androidx.lifecycle:lifecycle-extensions:2.0.0-rc01'
    implementation 'androidx.annotation:annotation:1.0.0'

Thanks,

Luce Luo
DJI Dev Team

neilyoung, 2019-09-25:
@dji-dev Does not compile with the original, adapted sample:

```
Executing tasks: [:app:assembleDebug]

:app:buildInfoDebugLoader
:app:preBuild UP-TO-DATE
:app:preDebugBuild UP-TO-DATE
:app:compileDebugAidl UP-TO-DATE
:app:compileDebugRenderscript UP-TO-DATE
:app:checkDebugManifest UP-TO-DATE
:app:generateDebugBuildConfig UP-TO-DATE
:app:prepareLintJar UP-TO-DATE
:app:mainApkListPersistenceDebug UP-TO-DATE
:app:generateDebugResValues UP-TO-DATE
:app:generateDebugResources UP-TO-DATE
:app:mergeDebugResources UP-TO-DATE
:app:createDebugCompatibleScreenManifests UP-TO-DATE
:app:processDebugManifest
:app:splitsDiscoveryTaskDebug UP-TO-DATE
:app:processDebugResources
:app:generateDebugSources
:app:javaPreCompileDebug
Note: /Users/decades/Documents/tmp/Android-VideoStreamDecodingSample/android-videostreamdecodingsample/app/src/main/java/com/dji/videostreamdecodingsample/media/DJIVideoStreamDecoder.java uses or overrides a deprecated API.
Note: Recompile with -Xlint:deprecation for details.
:app:compileDebugJavaWithJavac
:app:instantRunMainApkResourcesDebug
:app:mergeDebugShaders
:app:compileDebugShaders
:app:generateDebugAssets
:app:mergeDebugAssets
:app:validateSigningDebug
:app:processInstantRunDebugResourcesApk
:app:checkManifestChangesDebug
:app:transformClassesWithExtractJarsForDebug
:app:transformClassesWithInstantRunVerifierForDebug
:app:transformClassesWithDependencyCheckerForDebug UP-TO-DATE
:app:compileDebugNdk NO-SOURCE
:app:mergeDebugJniLibFolders
:app:transformNativeLibsWithMergeJniLibsForDebug
:app:processDebugJavaRes NO-SOURCE
:app:transformResourcesWithMergeJavaResForDebug
:app:transformNativeLibsAndResourcesWithJavaResourcesVerifierForDebug
:app:transformClassesWithInstantRunForDebug
:app:transformClassesEnhancedWithInstantReloadDexForDebug
:app:incrementalDebugTasks
:app:preColdswapDebug
:app:fastDeployDebugExtractor
:app:generateDebugInstantRunAppInfo
:app:transformClassesWithInstantRunSlicerForDebug
:app:transformClassesWithDexBuilderForDebug
AGPBI: {"kind":"error","text":"Program type already present: android.support.v4.os.ResultReceiver$1","sources":[{}],"tool":"D8"}
:app:transformDexArchiveWithExternalLibsDexMergerForDebug FAILED
:app:buildInfoGeneratorDebug

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':app:transformDexArchiveWithExternalLibsDexMergerForDebug'.
> com.android.builder.dexing.DexArchiveMergerException: Error while merging dex archives: /Users/decades/Documents/tmp/Android-VideoStreamDecodingSample/android-videostreamdecodingsample/app/build/intermediates/transforms/dexBuilder/debug/0.jar, /Users/decades/Documents/tmp/Android-VideoStreamDecodingSample/android-videostreamdecodingsample/app/build/intermediates/transforms/dexBuilder/debug/1.jar, /Users/decades/Documents/tmp/Android-VideoStreamDecodingSample/android-videostreamdecodingsample/app/build/intermediates/transforms/dexBuilder/debug/2.jar, /Users/decades/Documents/tmp/Android-VideoStreamDecodingSample/android-videostreamdecodingsample/app/build/intermediates/transforms/dexBuilder/debug/3.jar, /Users/decades/Documents/tmp/Android-VideoStreamDecodingSample/android-videostreamdecodingsample/app/build/intermediates/transforms/dexBuilder/debug/4.jar, /Users/decades/Documents/tmp/Android-VideoStreamDecodingSample/android-videostreamdecodingsample/app/build/intermediates/transforms/dexBuilder/debug/5.jar, /Users/decades/Documents/tmp/Android-VideoStreamDecodingSample/android-videostreamdecodingsample/app/build/intermediates/transforms/dexBuilder/debug/6.jar, /Users/decades/Documents/tmp/Android-VideoStreamDecodingSample/android-videostreamdecodingsample/app/build/intermediates/transforms/dexBuilder/debug/7.jar, /Users/decades/Documents/tmp/Android-VideoStreamDecodingSample/android-videostreamdecodingsample/app/build/intermediates/transforms/dexBuilder/debug/8.jar, /Users/decades/Documents/tmp/Android-VideoStreamDecodingSample/android-videostreamdecodingsample/app/build/intermediates/transforms/dexBuilder/debug/9.jar, /Users/decades/Documents/tmp/Android-VideoStreamDecodingSample/android-videostreamdecodingsample/app/build/intermediates/transforms/dexBuilder/debug/10.jar, /Users/decades/Documents/tmp/Android-VideoStreamDecodingSample/android-videostreamdecodingsample/app/build/intermediates/transforms/dexBuilder/debug/11.jar, /Users/decades/Documents/tmp/Android-VideoStreamDecodingSample/android-videostreamdecodingsample/app/build/intermediates/transforms/dexBuilder/debug/12.jar, /Users/decades/Documents/tmp/Android-VideoStreamDecodingSample/android-videostreamdecodingsample/app/build/intermediates/transforms/dexBuilder/debug/13.jar, /Users/decades/Documents/tmp/Android-VideoStreamDecodingSample/android-videostreamdecodingsample/app/build/intermediates/transforms/dexBuilder/debug/14.jar, /Users/decades/Documents/tmp/Android-VideoStreamDecodingSample/android-videostreamdecodingsample/app/build/intermediates/transforms/dexBuilder/debug/15.jar, /Users/decades/Documents/tmp/Android-VideoStreamDecodingSample/android-videostreamdecodingsample/app/build/intermediates/transforms/dexBuilder/debug/16.jar, /Users/decades/Documents/tmp/Android-VideoStreamDecodingSample/android-videostreamdecodingsample/app/build/intermediates/transforms/dexBuilder/debug/17.jar, /Users/decades/Documents/tmp/Android-VideoStreamDecodingSample/android-videostreamdecodingsample/app/build/intermediates/transforms/dexBuilder/debug/18.jar, /Users/decades/Documents/tmp/Android-VideoStreamDecodingSample/android-vi [... cut ...]
        exclude module: 'library-anti-distortion'
        exclude module: 'fly-safe-database'
    })
    compileOnly 'com.dji:dji-sdk-provided:4.11'
    implementation 'androidx.appcompat:appcompat:1.0.0'
    implementation 'androidx.core:core:1.0.0'
    implementation 'androidx.constraintlayout:constraintlayout:1.1.3'
    implementation 'androidx.recyclerview:recyclerview:1.0.0'
    implementation 'androidx.lifecycle:lifecycle-extensions:2.0.0-rc01'
    implementation 'androidx.annotation:annotation:1.0.0'

}

```

dji-dev, 2019-09-25:
Public comment from Luce Luo in <a href='https://djisdksupport.zendesk.com/agent/tickets/28278'>Zendesk ticket #28278</a>:<br><br>Dear Neil,

Thank you for contacting DJI.
Sorry, please wait for the official update for this sample.

Thanks,

Luce Luo
DJI Dev Team

cwbsl, 2019-10-01:
Any update on this issue? I'm facing the same problem and urgently need to resubmit the app to the play store before it get's removed because of the AMAP SDK.

neilyoung, 2019-10-01:
@cwbsl Could you please be a bit more specific? What "this" fixes what problem? Did you comment on the wrong thread?

neilyoung, 2019-10-01:
@dji-dev OK, I will let you participate.

You need to upgrade to gradle 3.5.0. Then right mouse click your project and `Refactor\Migrate to Android X'

This will at least produce something, which can be loaded to your device. If it runs? Tomorrow more...

cwbsl, 2019-10-02:
> @cwbsl Could you please be a bit more specific? What "this" fixes what problem? Did you comment on the wrong thread?

No, I'm completely in the right thread, the answer by dji-dev https://github.com/DJI-Mobile-SDK-Tutorials/Android-VideoStreamDecodingSample/issues/59#issuecomment-534923232
fixes the issue.
Make sure you have implementation 'androidx.lifecycle:lifecycle-extensions:2.0.0-rc01' in your build.gradle file.

neilyoung, 2019-10-02:
OK, I don't have that extension in build.gradle, but I could make it run following my receipt (one post above yours). Updating to gradle 3.2.0 (I was updating to 3.2.5) and "Refactoring to AndroidX" did it for the original sample.

closed 2019-10-08 by Michael-DJI