DJI-Mobile-SDK-Tutorials/Android-VideoStreamDecodingSample #60: Drone doesn't connect v 4.11

rabbishuki, 2019-11-06:
I updated the sample application to version 4.11 and it builds successfully.

Installed on my phone, plug in a P4A, and it doesn't connect.
A toast comes up saying "Disconnected".

I tried with Matrice 100 and got the same result.

Is the problem in your sample app? or does it have to do with Android x changes.

rabbishuki, 2019-11-06:
The problem was that updating the project deleted the SDK Key

closed 2019-11-06 by rabbishuki

dji-dev, 2019-11-07:
Public comment from Luce Luo in <a href='https://djisdksupport.zendesk.com/agent/tickets/29615'>Zendesk ticket #29615</a>:<br><br>Dear Customer,

Thank you for contacting DJI.
Great to hear that you have solved the problem by yourself.
Please feel free to contact us via the format once any problems.
https://formcrafts.com/a/dji-developer-feedback-en

Thanks,

Luce Luo
DJI Dev Team