DJI-Mobile-SDK-Tutorials/Android-VideoStreamDecodingSample #61: No implementation found for java.lang.String dji.midware.natives.SDKRelativeJNI.native_getUsbAccessoryAttachedString()

gsrathoreniks, 2020-02-25:
**Description of the issue.**
While launching the app on Emulator it crashes.

**Steps to reproduce the bug or crash issues(It would be great if you can provide this)**
Run the app on Android Emulator with configurations added below

Crash logs (If you can find it, it would be very helpful)

> 2020-02-25 19:58:35.190 19852-19870/? E/art: No implementation found for java.lang.String dji.midware.natives.SDKRelativeJNI.native_getUsbAccessoryAttachedString() (tried Java_dji_midware_natives_SDKRelativeJNI_native_1getUsbAccessoryAttachedString and Java_dji_midware_natives_SDKRelativeJNI_native_1getUsbAccessoryAttachedString__)
> 2020-02-25 19:58:35.191 19852-19852/? E/AndroidRuntime: FATAL EXCEPTION: main
>     Process: com.dji.videostreamdecodingsample, PID: 19852
>     java.lang.NoClassDefFoundError: <clinit> failed for class dji.sdk.sdkmanager.DJISDKManager; see exception in other thread
>         at dji.sdk.sdkmanager.DJISDKManager.getInstance()
>         at com.dji.videostreamdecodingsample.ConnectionActivity.initUI(Unknown Source)
>         at com.dji.videostreamdecodingsample.ConnectionActivity.onCreate(Unknown Source)
>         at android.app.Activity.performCreate(Activity.java:6679)
>         at android.app.Instrumentation.callActivityOnCreate(Instrumentation.java:1118)
>         at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2618)
>         at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:2726)
>         at android.app.ActivityThread.-wrap12(ActivityThread.java)
>         at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1477)
>         at android.os.Handler.dispatchMessage(Handler.java:102)
>         at android.os.Looper.loop(Looper.java:154)
>         at android.app.ActivityThread.main(ActivityThread.java:6119)
>         at java.lang.reflect.Method.invoke(Native Method)
>         at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:886)

**DJI Android SDK version you are using (Like Android SDK 3.4, etc)**
Android SDK 4.11.1

**DJI Product you are using (Like Phantom 4, Mavic Pro, etc)**
Mavic 2 Pro

**Android system version you are using (Like Android 6.0.1, Android 5.1.1, etc)**
Android 7.1.1

**Android device you are using (Like Samsung Galaxy Note 5, Nexus 5, etc)**
Pixel 2 (API 25) Android Emulator

**Android Studio version you are using (Like Android Studio 2.2, etc)**
Android Studio 3.5.3

dji-dev, 2020-02-25:
Public comment from Lisa Fedane in <a href='https://djisdksupport.zendesk.com/agent/tickets/32364'>Zendesk ticket #32364</a>:<br><br>What ABI are you using for the virtual device?  DJI's SDK does not support x86 devices, which is what Android Studio recommends for its virtual devices.  You can choose another setting, but it takes a very long time to build and when your app is not connected to a DJI hardware product, there is very limited development that you can do - best practice is to just run the DJI SDK based app on a real Android device.

See Release Notes here: https://developer.dji.com/mobile-sdk/downloads/
 [image]​

For testing purposes, we recommended using the Android Bridge App: https://github.com/dji-sdk/Android-Bridge-App

Lisa Fedane
DJI Developer Support<br><a href='https://djisdksupport.zendesk.com/attachments/token/0dAm17Yt2YrB3AOrxRr3um0Ji' target='zd_attach'>inline-1163597199.png</a>

gsrathoreniks, 2020-02-26:
I've tried running on 
**armeabi-v7a
x86
x86_64**

I'm running Android Bridge App on an Android device that is connected to RC, on the other hand, I'm running VideoStreamDecodingSample on Android Emulator and here on the emulator it's crashing.

What ABI do you recommend me to use?

dji-dev, 2020-02-26:
Public comment from Lisa Fedane in <a href='https://djisdksupport.zendesk.com/agent/tickets/32364'>Zendesk ticket #32364</a>:<br><br>We do not officially recommend that you use Android Studio emulator

Lisa Fedane
DJI Developer Support

gsrathoreniks, 2020-02-26:
@dji-dev  Yeah, I got it. But there has to be some way to get this done right?
As the Android Bridge App is of no use, it was built so that one can use the SDK/VideoDecoding apps on Emulators.

dji-dev, 2020-02-26:
Public comment from Lisa Fedane in <a href='https://djisdksupport.zendesk.com/agent/tickets/32364'>Zendesk ticket #32364</a>:<br><br>I think there maybe an misunderstanding - the bridge app is meant to be used with two real Android devices:
  [image]​

Lisa Fedane
DJI Developer Support<br><a href='https://djisdksupport.zendesk.com/attachments/token/hfVRx8ZmxEuEQjfAvckVgrSE5' target='zd_attach'>inline-2055421612.png</a>

gsrathoreniks, 2020-02-26:
So, this means there's no way to get this done!

dji-dev, 2020-02-26:
Public comment from Lisa Fedane in <a href='https://djisdksupport.zendesk.com/agent/tickets/32364'>Zendesk ticket #32364</a>:<br><br>**Although we recommend not using the emulator the MSDK does supports the following abi filters:**

// On x86 devices that run Android API 23 or above, if the application is targeted with API 23 or
// above, FFmpeg lib might lead to runtime crashes or warnings.
abiFilters 'armeabi-v7a', 'x86', 'arm64-v8a'

So yes, if you get an emulator working, then it _should_ be able to connect to the bridge app as long as both the bridge app device and emulator are in the same network.

What happened when you ran the emulator on armeabi-v7a?  You got the same crash?

Lisa Fedane
DJI Developer Support

gsrathoreniks, 2020-02-27:
Yes, It got the same crash.