DJI-Mobile-SDK-Tutorials/Android-VideoStreamDecodingSample #64: Should not use milliseconds as pts in

msunshinelxl, 2020-06-13:


msunshinelxl, 2020-06-13:
In the [file](https://github.com/DJI-Mobile-SDK-Tutorials/Android-VideoStreamDecodingSample/blob/master/android-videostreamdecodingsample/app/src/main/java/com/dji/videostreamdecodingsample/media/DJIVideoStreamDecoder.java), onDataRecv func, DJIFrame's pts should be microseconds instead of milliseconds.

dji-dev, 2020-06-15:
Agent comment from Luce Luo in <a href='https://djisdksupport.zendesk.com/agent/tickets/35766'>Zendesk ticket #35766</a>:<br><br>Dear Customer,

Thank you for contacting DJI.
To help us collect questions in detail, please fill in the form below and we will handle this ticket once we have received it.
https://formcrafts.com/a/dji-developer-feedback-en

Thanks,

Luce Luo
DJI Dev Team