DJI-Mobile-SDK-Tutorials/Android-VideoStreamDecodingSample #65: 解码成功但是数据量错误

mudcastles, 2020-08-17:
按照demo的代码复制到自己的项目中，在御2行业版飞机上对H264源流解码，得到的bufferInfo的offset是0，size是2,402,304。儿分辨率是1440*1088。按照此分辨率计算，size应为2,350,080。难道是得到的bufferInfo的offset值错了？

dji-dev, 2020-08-18:
Agent comment from William Wong in <a href='https://djisdksupport.zendesk.com/agent/tickets/38086'>Zendesk ticket #38086</a>:<br><br>尊敬的开发者，
您好，感谢您联系DJI 大疆创新。

应该不会有错，不然图传不能够正常显示的。

希望我们的解决方案能够帮到您，感谢您的邮件，祝您生活愉快！
Best Regards,
DJI 大疆创新SDK技术支持

mudcastles, 2020-08-21:
> Agent comment from William Wong in [Zendesk ticket #38086](https://djisdksupport.zendesk.com/agent/tickets/38086):
> 
> 尊敬的开发者，
> 您好，感谢您联系DJI 大疆创新。
> 
> 应该不会有错，不然图传不能够正常显示的。
> 
> 希望我们的解决方案能够帮到您，感谢您的邮件，祝您生活愉快！
> Best Regards,
> DJI 大疆创新SDK技术支持

您好，我是使用的DJIVideoStreamDecoder进行解码得到的上述结果。此解码的用途是合成jpeg图片。而且我重新启动了遥控器和飞机后，这个问题不能复现了，但是这个问题确实出现过。