DJI-Mobile-SDK-Tutorials/Android-VideoStreamDecodingSample #69: mavic mini compatibility

lukelu0520, 2020-11-15:
Dear DJI Mobile SDK VideoStreamDecoding friends,

As we know, version 4.13.1 starts supporting Mavic Mini ! So we'd like to do video stream decoding as people did for Mavic 2 Pro or Zoom.

However, we found there is an issue on Mini's decoding, we can not get ffmpeg parser correctly, resulting bad video stream.

We were wondering if anyone is able to make this github working on Mavic Mini. Hope to hear your expertise.

Best,
-- Luke

lukelu0520, 2020-11-16:
Update: the answer is yes !

neilyoung, 2020-11-16:
Strange thread. Does that mean, the issue is gone or will be fixed soon?

lukelu0520, 2020-11-16:
We are able to see video stream from Mini's camera on android device. But we are working on streaming the video via rtp / udp (e.g. in H264). If you know any way to make it work, please let us know.

Hope to hear your expertise.

Best,
-- Luke

neilyoung, 2020-11-16:
Yes. I know and have a better way already worked out since a year and more. WebRTC. If you want to know details please mail me at decades-software at freenet dot de.

neilyoung, 2020-11-16:
Just a quick question: Could you please try if this works for you on a  Mini and an Android phone? This is what I have done, but I don't have a Mini

https://github.com/accuware/djistreamerlib

lukelu0520, 2020-11-17:
@neilyoung , thanks for your information. 
As a feedback, we found your app does not recognize Mini. So we even can not press the OPEN button to start with.
Hope to hear your solution.

Best,
-- Luke

neilyoung, 2020-11-17:
OK, sorry to hear that. This can of course have a lot of reasons. I don't think the lib is too chatty, but is there anything in the ADB log, which looks suspecious? Then: Did you apply for the test credentials on our website and configure this for your use?

In a first step I could try to add a bit more debug information, maybe this helps. 

As a further option I think it would make sense for me to lean such a device for a month or so in order to debug the problem. Let me know, what exact model is your preferred SUT.

https://www.grover.com/de-de/drones/for-pro?adwords_campaign=sitelink_extension&filter=in_stock%3Dtrue%26rental_plan%3D1&gclid=Cj0KCQiA48j9BRC-ARIsAMQu3WSe5PIOZmGNno0sDIatUEtM82QDW1Q3WF7H22LBcRoH4dXXi6dc4F4aAi3tEALw_wcB&utm_campaign=S-DE%7Cct%3ANBP%7Cbr%3Adji%7Cty%3AGeneric&utm_medium=cpc&utm_source=google&utm_term=mavic+mini

neilyoung, 2020-11-17:
BTW: Since I'm willing to debug this problem with you (if possible), are you kind of adventurous too? If so then please provide me a Dropbox or kind of way to upload more chatty versions of the lib. Thanks

neilyoung, 2020-11-17:
Or maybe we should continue by mail. If so, please ping me first.

neilyoung, 2020-11-17:
> As a feedback, we found your app does not recognize Mini. So we even can not press the OPEN button to start with.
> Hope to hear your solution.
> 
> Best,
> -- Luke

BTW: Could you clarify, what OPEN button you are unable to press?

What I would expect you to see is:

1) You connect the Android device with "my" (which is basically your's) VideoStreamDecodingSample installed with the Remote Controller
2) On the Android device there is popping up a box, asking you, what software should be used with the plugged controller. You have the choice between DJIGO4 and "Video Stream Decoding Sample". You choose the latter, for JUST ONCE
3) You enter a product information screen. If the "OPEN" button there is not getting blue over time and there is no product and software version information, already the first authentication steps with your platform did fail. You have not even reached the level of the real application.

Did you configure a valid DJI Authentication Key in AndroidManifest.xml?

neilyoung, 2020-11-18:
Any feedback so far?

lukelu0520, 2020-11-19:
We are using MAVIC 2 ZOOM to see if it works. Also, we can not see MINI listed in your [source code](https://github.com/accuware/android-videostreamdecodingsample-webrtc/blob/master/app/src/main/java/com/dji/videostreamdecodingsample/media/DJIVideoStreamDecoder.java). This might be the root cause of the connectivity issue on MINI. Hope you could double confirm it.

Seems like our discussion goes beyond this github. I'll close the issue here, and file issue on your github or contact you via your email for future questions.

Thank you !
-- Luke

closed 2020-11-19 by lukelu0520

neilyoung, 2020-11-19:
Hi Luke, yes. I‘m fine with this. But please note (I’m sure you know that): What you call „your code“ is basically your code 😅. I’m totally convinced that the Mini is not part of any enumeration there since the sample has not been aligned with your probably updated latest sample code. I can do that in a blind flight, but since you are the original creators of this code it should be easy for you to PR me a change to move forward. Agreed?

neilyoung, 2020-11-19:
> Thanks for your active support! We are using MAVIC 2 ZOOM to see if it works. Then we'll know if it is a special problem to MINI. Also, seems like our discussion goes beyond this github. I'll close the issue here, and file issue on your github or contact you via your email in the future.
> 
> Best,
> -- Luke

You have removed this passage from your original post:

> Also, we can not see MINI listed in your source code. This might be the root cause of the connectivity issue on MINI. Hope you could double confirm it.

Well, I cannot see this in your latest version of the sample either https://github.com/DJI-Mobile-SDK-Tutorials/Android-VideoStreamDecodingSample/blob/master/android-videostreamdecodingsample/app/src/main/java/com/dji/videostreamdecodingsample/media/DJIVideoStreamDecoder.java 

So if you don't have it, how can I have it? :)

neilyoung, 2020-11-20:
Anyway, I will update "my" sample app so that it reflects the current state and from that we can look further.

accuware, 2020-11-20:
Sorry, comment above from another account (but still valid :))

OK, I have updated djistreamerlib (0.8.0 now) and sample app (also 0.8.0) to reflect the requirements of SDK 4.13.1 https://github.com/accuware/android-videostreamdecodingsample-webrtc

I still see no trace of a special treatment of the DJI Mini drone. I have tested the functionality of my lib and the sample app with a fresh build of both here on my Mavic Pro and Android 8 (Google device). It works fine. Please re-test and probably let me know, what to do with the Mini in particular.

neilyoung, 2020-11-20:
OK, I have updated djistreamerlib (0.8.0 now) and sample app (also 0.8.0) to reflect the requirements of SDK 4.13.1 https://github.com/accuware/android-videostreamdecodingsample-webrtc

I still see no trace of a special treatment of the DJI Mini drone. I have tested the functionality of my lib and the sample app with a fresh build of both here on my Mavic Pro and Android 8 (Google device). It works fine. Please re-test and probably let me know, what to do with the Mini in particular.

neilyoung, 2020-11-23:
@lukelu0520 You should quickly decide if you need my assistance in this matter. The company is going OOB and I will soon remove the code from the web. I have aligned to your 4.13.1 SDK. Let me know, I will wait till EOB this week and then shutdown everything.