DJI-Mobile-SDK-Tutorials/Android-VideoStreamDecodingSample #70: Is it now ready for 4.13 or still working with 4.11?

neilyoung, 2020-11-19:
Hi, according to several comments in the commit history this sample has been aligned to run with 4.13 SDK. In contrary the build.gradle still makes use of the 4.11 SDK. What am I missing?

https://github.com/DJI-Mobile-SDK-Tutorials/Android-VideoStreamDecodingSample/blob/888ae23dd90e119999a05a5d7085e2f74f21259d/android-videostreamdecodingsample/app/build.gradle#L77-L87

dji-dev, 2020-11-20:
Agent comment from Luce Luo in <a href='https://djisdksupport.zendesk.com/agent/tickets/41252'>Zendesk ticket #41252</a>:<br><br>Dear Customer,

Thank you for contacting DJI.
In fact, the Gradle will be configured by the last script that will apply the 4.13.1 configuration.
 [image]​
https://github.com/DJI-Mobile-SDK-Tutorials/Android-VideoStreamDecodingSample/blob/888ae23dd90e119999a05a5d7085e2f74f21259d/android-videostreamdecodingsample/app/build.gradle#L99

Thanks,

Luce Luo
DJI Developer Support<br><a href='https://djisdksupport.zendesk.com/attachments/token/oirYK6qCttjneOHcJ55HL8LVF' target='zd_attach'>inline-1699188551.png</a>

neilyoung, 2020-11-20:
Ah, I see. Grrr... very weird roundabout. Is that really necessary? Anyway, I see, that it finally applies 4.13.

OK, back to "the other thing". In another thread (https://github.com/DJI-Mobile-SDK-Tutorials/Android-VideoStreamDecodingSample/issues/69#issuecomment-730078898) you where mentioning, that "my" code would not support the mini (the comment has later been altered, I just have it on mail), since it is not mentioned in DJIVideoStreamDecoder.java. I cannot find any mentioning of this drone in this project as well, so what about this?

dji-dev, 2020-11-20:
Agent comment from Luce Luo in <a href='https://djisdksupport.zendesk.com/agent/tickets/41252'>Zendesk ticket #41252</a>:<br><br>Dear Customer,

Thank you for contacting DJI.
Sorry, but I'm not Luke who you ping on issue69 thread.
If you have other SDK questions, it's recommended to contact us via the form.
https://formcrafts.com/a/dji-developer-feedback-cn

Thanks,

Luce Luo
DJI Developer Support

neilyoung, 2020-11-20:
Ah, ok. Disregard please. My bad.