DJI-Mobile-SDK-Tutorials/Android-VideoStreamDecodingSample #72: H.264 framerate

neilyoung, 2021-04-06:
Hi, 

I'm having a question regarding the frame rate and format of the H.264 stream coming from the SDK via `VideoFeeder.VideoDataListener()`.  From the SPS packets in the H.264 stream it looks like variable frame rate:

```
[trace_headers @ 0x7fc69d505200] 108         timing_info_present_flag                                    1 = 1
[trace_headers @ 0x7fc69d505200] 109         num_units_in_tick            00000000000000000000000000000001 = 1
[trace_headers @ 0x7fc69d505200] 141         time_scale                   00000000000000000000000000111100 = 60
[trace_headers @ 0x7fc69d505200] 173         fixed_frame_rate_flag                                       0 = 0

```

However, I can't get more out of it than 20 fps at 1280 x 720, forwarding the H.264 directly via RTP/RTSP. Any information?

dji-dev, 2021-04-07:
Agent comment from Luce Luo in <a href='https://djisdksupport.zendesk.com/agent/tickets/46055'>Zendesk ticket #46055</a>:<br><br>Dear Customer,

Thank you for contacting DJI.
The SDK doesn't support the RTP/RTSP, if you are going to use the live stream.
Please use the RTMP LiveStreamManager.
https://developer.dji.com/api-reference/android-api/Components/LiveStreamManager/DJILiveStreamManager.html
And the resolution is 720P/30fps default, some drone support the 1080p with setHDLiveViewEnabled method.

Thanks,

Luce Luo
DJI Developer Support

neilyoung, 2021-04-07:
> The SDK doesn't support the RTP/RTSP, if you are going to use the live stream.
I know, that's why I'm taking the H.264 as it comes from the drone and wrap it with the RTP/RTSP stuff, providing my own RTSP server on the Android device. However, I can't get more than 20 fps on the reception side for 1280 x 720. How can you be sure, your source delivers 30 fps?

dji-dev, 2021-04-08:
Agent comment from Luce Luo in <a href='https://djisdksupport.zendesk.com/agent/tickets/46055'>Zendesk ticket #46055</a>:<br><br>Dear Customer,

Thank you for contacting DJI.
I see.
The fps is controlled by the camera, which will be determined by the current view.
It's also possible to transmit with a speed lower than 30fps if the current view is stable.

Thanks,

Luce Luo
DJI Developer Support

neilyoung, 2021-04-08:
This seems to be happening here. But unfortunately I'm completely unable to make it send with 30 fps. Isn't there any trick, any setting, any API to achieve this?

dji-dev, 2021-04-09:
Agent comment from Luce Luo in <a href='https://djisdksupport.zendesk.com/agent/tickets/46055'>Zendesk ticket #46055</a>:<br><br>Dear Customer,

Thank you for contacting DJI.
Sorry that there is no more setting to adjust it, it comes from the camera.

Thanks,

Luce Luo
DJI Developer Support

neilyoung, 2021-04-09:
You mean the H.264 comes from the camera? Doubt that TBH

dji-dev, 2021-04-09:
Agent comment from Luce Luo in <a href='https://djisdksupport.zendesk.com/agent/tickets/46055'>Zendesk ticket #46055</a>:<br><br>Dear Customer,

Thank you for contacting DJI.
Yes, sorry that I can't give more help with this problem.
There is no SDK way to control the FPS of the live view.

Thanks,

Luce Luo
DJI Developer Support

neilyoung, 2021-04-09:
I was not necessarily after an API, I was after a confirmation, that 20 fps is ok..

Look, this is my theory: You are getting the video as plain YUV or RGB uncompressed from the drone via your wireless channel. Then the SDK utilizes the H.264 encoding caps of the underlaying Android system and encodes (and decodes again via the yuv callback) the frames. In this scenario I would understand if the framerate is 20 fps and lower at 1280x720, since I also cannot achieve more with my means here. But this would at least give the chance to see > 30fps on other, more capable devices (!?)

dji-dev, 2021-04-12:
Agent comment from Luce Luo in <a href='https://djisdksupport.zendesk.com/agent/tickets/46055'>Zendesk ticket #46055</a>:<br><br>Dear Customer,

Thank you for contacting DJI.
Sorry about it.
I will record your advice and recommended using the official LiveStream feature for the being time.

Thanks,

Luce Luo
DJI Developer Support