DJI-Mobile-SDK-Tutorials/Android-VideoStreamDecodingSample #73: Re-Compiling arm64-v8a doesnt work

enoWal, 2021-05-18:
Hi! I got a problem with recompiling the libs of this projekt.

**Information:**
Which DJI Product you are using?
-Mavic 2 Enterprise Zoom
Which Android Device and Android system version you are using?
-LG G7 with Android 10
Which Android Studio version you are using?
-4.1.3

**A short description of my problem including terminal logs:**

Im trying to re-compile the libs with the android ndk.
I deleted all existing libs and obj in the folder directory and startet to recompile them with the jni folder and the ndk.

My problem is that only the armeabi-v7a and x86 are recompiled but the arm64-v8a isnt; so i cant use the lib in my own project.

I didnt change anything in the project except the App Key for testing.

I get this output in the terminal when i try to rebuild the lib.

```
C:\Users\moritz\Android-VideoStreamDecodingSample\android-videostreamdecodingsample>C:\Users\moritz\AppData\Local\Android\Sdk\ndk\22.1.7171670/ndk-build
fcntl(): Bad file descriptor
C:/Users/moritz/AppData/Local/Android/Sdk/ndk/22.1.7171670/build//../build/core/build-binary.mk:751: warning: overriding recipe for target 'obj/local/armeabi-v7a/libffmpeg.so'
C:/Users/moritz/AppData/Local/Android/Sdk/ndk/22.1.7171670/build//../build/core/build-binary.mk:751: warning: ignoring old recipe for target 'obj/local/armeabi-v7a/libffmpeg.so'
C:/Users/moritz/AppData/Local/Android/Sdk/ndk/22.1.7171670/build//../build/core/build-binary.mk:775: warning: overriding recipe for target 'libs/armeabi-v7a/libffmpeg.so'
C:/Users/moritz/AppData/Local/Android/Sdk/ndk/22.1.7171670/build//../build/core/build-binary.mk:775: warning: ignoring old recipe for target 'libs/armeabi-v7a/libffmpeg.so'
C:/Users/moritz/AppData/Local/Android/Sdk/ndk/22.1.7171670/build//../build/core/build-binary.mk:751: warning: overriding recipe for target 'obj/local/x86/libffmpeg.so'
C:/Users/moritz/AppData/Local/Android/Sdk/ndk/22.1.7171670/build//../build/core/build-binary.mk:751: warning: ignoring old recipe for target 'obj/local/x86/libffmpeg.so'
C:/Users/moritz/AppData/Local/Android/Sdk/ndk/22.1.7171670/build//../build/core/build-binary.mk:775: warning: overriding recipe for target 'libs/x86/libffmpeg.so'
C:/Users/moritz/AppData/Local/Android/Sdk/ndk/22.1.7171670/build//../build/core/build-binary.mk:775: warning: ignoring old recipe for target 'libs/x86/libffmpeg.so'
[armeabi-v7a] Install        : libdjivideojni.so => libs/armeabi-v7a/libdjivideojni.so
[armeabi-v7a] Install        : libffmpeg.so => libs/armeabi-v7a/libffmpeg.so
[armeabi-v7a] Install        : libyuv2.so => libs/armeabi-v7a/libyuv2.so
[x86] Install        : libdjivideojni.so => libs/x86/libdjivideojni.so
[x86] Install        : libffmpeg.so => libs/x86/libffmpeg.so
[x86] Install        : libyuv2.so => libs/x86/libyuv2.so
```
As you cant see it builds the 2 instruction sets, but the arm64 ist missing.

dji-dev, 2021-05-19:
Agent comment from Luce Luo in <a href='https://djisdksupport.zendesk.com/agent/tickets/47885'>Zendesk ticket #47885</a>:<br><br>Dear Customer,

Thank you for contacting DJI.
Sorry that the Android-VideoStreamDecodingSample is not recommended to recompile, please use the existing libs.

Thanks,

Luce Luo
DJI Developer Support
<br><p><em><small>°°°</small></em></p>

TaylorsZ, 2021-12-06:
> Agent comment from Luce Luo in [Zendesk ticket #47885](https://djisdksupport.zendesk.com/agent/tickets/47885):Dear Customer,
> 
> Thank you for contacting DJI. Sorry that the Android-VideoStreamDecodingSample is not recommended to recompile, please use the existing libs.
> 
> Thanks,
> 
> Luce Luo DJI Developer Support
> 
> _°°°_
我也遇到同样的问题，我在Application.mk里面添加APP_ABI := armeabi-v7a x86 arm64-v8a，编译完成后，使用M300+H20T视频无法预览(黑屏，删掉生成的arm64-v8a文件夹可以正常显示)，但因为项目的原因需要使用arm64，请问我该如何生成arm64的.so?

Acipenser, 2021-12-08:
Did you try to execute the `ndk-build` command with the `NDK_APPLICATION_MK="jni/Application64.mk"` param?

TaylorsZ, 2021-12-09:
> Did you try to execute the `ndk-build` command with the `NDK_APPLICATION_MK="jni/Application64.mk"` param?

感谢指导，加入这个参数后生成了arm64-v8a文件夹，但在M300下依旧黑屏。下载Android-VideoStreamDecodingSample后填写dji key， 使用H:\Android\sdk\ndk\23.0.7599858\ndk-build NDK_APPLICATION_MK="jni/Application64.mk"命令重编译后，demo里面的Demo TextureView和Demo SurfaceView均黑屏，但Demo custom decoder是正常的。另外使用精灵4p所有模式是可以正常显示的！ 请问我该如何操作？

sheng930920, 2022-09-28:
@TaylorsZ 咨询一下这个问题你解决了么，我也遇到同样的问题，重新编译 arm64-v8a，编译完成后，在御2行业进阶版无人机上使用视频无法预览黑屏