DJI-Mobile-SDK-Tutorials/Android-VideoStreamDecodingSample #75: M300 遥控器上解码花屏，编解码

stacker10, 2021-08-13:
初始化MediaCodec时，设置surface为null
codec.configure(format, null, null, 0);
在解码的时候，获取一帧数据
Image image = codec.getOutputImage(outIndex);
将这帧image存为图片

同样的代码在华为Mate30手机上面可以正常存储显示，在M300遥控器上面保存的这帧数据出现花屏。

请问M300遥控器硬件是否支持硬解码，或者哪些编解码格式，输出数据类型是什么

dji-dev, 2021-08-16:
Agent comment from William Wong in <a href='https://djisdksupport.zendesk.com/agent/tickets/51720'>Zendesk ticket #51720</a>:<br><br>最近也有其他用户提出这个问题，是否有时候从遥控器解码出来的yuv数据会花屏，但是同样的程序从手机里走出来就正常？我们正在定位这个问题，初步怀疑是智能遥控器的问题。
<br><p><em><small>°°°</small></em></p>

dji-dev, 2021-11-19:
Agent comment from William Wong in <a href='https://djisdksupport.zendesk.com/agent/tickets/51720'>Zendesk ticket #51720</a>:<br><br>这个问题解决了，4.16版本就会恢复正常。
<br><p><em><small>°°°</small></em></p>