DJI-Mobile-SDK-Tutorials/Android-VideoStreamDecodingSample #77: Does not compile with latest Android Studio, Mac OS

neilyoung, 2022-01-31:
Was trying to compile this as is with Android Studio Bumblebee. Does not work

```
2022-01-31T18:48:02.360+0100 [ERROR] [org.gradle.internal.buildevents.BuildExceptionReporter] 
2022-01-31T18:48:02.360+0100 [ERROR] [org.gradle.internal.buildevents.BuildExceptionReporter] FAILURE: Build failed with an exception.
2022-01-31T18:48:02.360+0100 [ERROR] [org.gradle.internal.buildevents.BuildExceptionReporter] 
2022-01-31T18:48:02.360+0100 [ERROR] [org.gradle.internal.buildevents.BuildExceptionReporter] * What went wrong:
2022-01-31T18:48:02.360+0100 [ERROR] [org.gradle.internal.buildevents.BuildExceptionReporter] Execution failed for task ':app:processDebugMainManifest'.
2022-01-31T18:48:02.360+0100 [ERROR] [org.gradle.internal.buildevents.BuildExceptionReporter] > Unable to make field private final java.lang.String java.io.File.path accessible: module java.base does not "opens java.io" to unnamed module @1223287
2
```

Most likely because of this in AndroidManifest.xml:

<uses-permission android:name="android.permission.MOUNT_UNMOUNT_FILESYSTEMS"/>

At least commenting it helps

dji-dev, 2022-01-31:
Agent comment from DJI SDK in <a href='https://djisdksupport.zendesk.com/agent/tickets/59854'>Zendesk ticket #59854</a>:<br><br>尊敬的开发者，感谢您联系DJI 大疆创新
由于github不是我们主要的咨询渠道，您的问题可能跟进不及时。我们建议您通过填写表单（ https://djisdksupport.zendesk.com/hc/zh-cn/requests/new ）向我们反馈问题。或者您也可以在论坛发帖，与其它开发者交流。论坛链接：https://djisdksupport.zendesk.com/hc/zh-cn/community/topics

Dear developer, thank you for contacting DJI.
Since github is not our main consultation channel, your questions may not be followed up in time. We recommend that you fill in the form (https://djisdksupport.zendesk.com/hc/en-us/requests/new) to report problems to us. Or you can post in the forum to communicate with other developers. Forum link: https://djisdksupport.zendesk.com/hc/zh-cn/community/topics
<br><p><em><small>°°°</small></em></p>

neilyoung, 2022-01-31:
Release build fails with:

```
> Task :app:lintVitalRelease FAILED
/Users/neilyoung/StudioProjects/original-dji-sample/android-videostreamdecodingsample/app/src/main/res/layout/activity_main.xml:94: Error: @id/activity_main_screen_shot is not a sibling in the same RelativeLayout [NotSibling]
        android:layout_toRightOf="@id/activity_main_screen_shot"
        ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

   Explanation for issues of type "NotSibling":
   Layout constraints in a given ConstraintLayout or RelativeLayout should
   reference other views within the same relative layout (but not itself!)

1 errors, 0 warnings

Execution failed for task ':app:lintVitalRelease'.
> Lint found fatal errors while assembling a release target.

```

DJI-William, 2022-02-02:
Have you modified anything in the project? It can compile at my side, please look at your tool version.
classpath 'com.android.tools.build:gradle:4.2.1'
distributionUrl=https\://services.gradle.org/distributions/gradle-7.1-all.zip

neilyoung, 2022-02-02:
Not a single bit. Settings are identical.

neilyoung, 2022-02-02:
Repeated the procedure. Slightly different result. Debug build ok, but trying to generate signed release apk fails repeatedly with:

`10:40	Generate Signed APK: Errors while building APK. You can find the errors in the 'Messages' view.
`

Message view:

```
Execution failed for task ':app:lintVitalRelease'.
> Lint found fatal errors while assembling a release target.

```

In Android Manifest this line is underlined red:

`    <uses-permission android:name="android.permission.MOUNT_UNMOUNT_FILESYSTEMS"/>
`

The warning tool suggest to add "Suppress add tools:ignore="ProtectPermissions" attribute.

The error is not disappearing if commenting this line.

`./gradlew clean assembleRelease --info
`
```
:app:extractDeepLinksRelease (Thread[Execution worker for ':' Thread 4,5,main]) completed. Took 0.002 secs.
:app:processReleaseMainManifest (Thread[Execution worker for ':' Thread 4,5,main]) started.

> Task :app:processReleaseMainManifest FAILED
Caching disabled for task ':app:processReleaseMainManifest' because:
  Build cache is disabled
Task ':app:processReleaseMainManifest' is not up-to-date because:
  Value of input property 'variantOutput.outputFileName' has changed for task ':app:processReleaseMainManifest'
The input changes require a full rebuild for incremental task ':app:processReleaseMainManifest'.
Unable do incremental execution: full task run

```

Can only get rid of this with the lintOption above.

Android Studio:

[image]

[image]

Cloned the project like so:

`git clone https://github.com/DJI-Mobile-SDK-Tutorials/Android-VideoStreamDecodingSample.git original-dji-sample
`
and opened it from Android Studio as new project from `original-dji-sample/android-videostreamdecodingsample/`

Any questions?

neilyoung, 2022-02-02:
Trying to build an apk works with the commented changes. Not trying to build an app bundle. 

```
10:54	Generate Signed Bundle: Errors while building Bundle file. You can find the errors in the 'Messages' view.

Version code not found in manifest.

```

IMHO you can't upload a new app to the store since November anymore, which is not packed as app bundle. Welcome to 2022.

DJI-William, 2022-02-04:
Change the MSDK version to 4.16 in build.gradle file and then click Build->Build Bundle(s)/APK(s)->Build Bundle(s), it will work.

neilyoung, 2022-02-04:
No, versionCode helps definitely. No need to change MSDK. But this is not the main issue. The main issue is, that it doesn't built as is as retail version. I thought you would be thankful to hear from that. But you are free to ignore.

closed 2022-02-04 by neilyoung