DJI-Mobile-SDK-Tutorials/Android-VideoStreamDecodingSample #78: How to solve a long term ambiguity in 2022?

neilyoung, 2022-02-04:
I already raised this issue a couple of years ago: The ambiguity of internal YUV format of buffers delivered by `onYuvDataReceived`.

https://github.com/dji-sdk/Mobile-SDK-Android/issues/422

I understood that this is mainly to be blamed on Android, which introduced this mess. I noticed that you addressed this with some changes in your latest sample and you were also adding this `format` parameter during the SDK developments:

```
            AsyncTask.execute(new Runnable() {
                @Override
                public void run() {
                    // two samples here, it may has other color format.
                    int colorFormat = format.getInteger(MediaFormat.KEY_COLOR_FORMAT);
                    switch (colorFormat) {
                        case MediaCodecInfo.CodecCapabilities.COLOR_FormatYUV420SemiPlanar:
                            //NV12
                            if (Build.VERSION.SDK_INT <= 23) {
                                oldSaveYuvDataToJPEG(bytes, width, height);
                            } else {
                                newSaveYuvDataToJPEG(bytes, width, height);
                            }
                            break;
                        case MediaCodecInfo.CodecCapabilities.COLOR_FormatYUV420Planar:
                            //YUV420P
                            newSaveYuvDataToJPEG420P(bytes, width, height);
                            break;
                        default:
                            break;
                    }

                }
            });
```

However, this solution jumps to short, since the used `MediaCodecInfo.CodecCapabilities.COLOR_FormatXXX` constants are all deprecated and the replacement (COLOR_FormatYUV420Flexible) embraces them all, which in turn hands out the ambiguity to the user: Even if a buffer is formatted as COLOR_FormatYUV420Flexible it can contain NV12, NV21, YV12 or I420P.

What to do with this?

dji-dev, 2022-02-04:
Agent comment from DJI SDK in <a href='https://djisdksupport.zendesk.com/agent/tickets/59909'>Zendesk ticket #59909</a>:<br><br>尊敬的开发者，感谢您联系DJI 大疆创新
由于github不是我们主要的咨询渠道，您的问题可能跟进不及时。我们建议您通过填写表单（ https://djisdksupport.zendesk.com/hc/zh-cn/requests/new ）向我们反馈问题。或者您也可以在论坛发帖，与其它开发者交流。论坛链接：https://djisdksupport.zendesk.com/hc/zh-cn/community/topics

Dear developer, thank you for contacting DJI.
Since github is not our main consultation channel, your questions may not be followed up in time. We recommend that you fill in the form (https://djisdksupport.zendesk.com/hc/en-us/requests/new) to report problems to us. Or you can post in the forum to communicate with other developers. Forum link: https://djisdksupport.zendesk.com/hc/zh-cn/community/topics
<br><p><em><small>°°°</small></em></p>

DJI-William, 2022-02-04:
We haven't maintain this demo for a long time because we are working on the SDK development. We will release a barnd new SDK architechture MSDK 5.0 very soon and we will give a full/comlete demo later. MSDK 4.X will be rarely updated anymore.

neilyoung, 2022-02-04:
Thanks for the quick info.