DJI-Mobile-SDK-Tutorials/Android-VideoStreamDecodingSample #79: MSDK 4.16 or DJI Mini 2 Support?

Waffle1434, 2022-03-09:
I don't see the mini 2 model mentioned in [DJIVideoStreamDecoder.java](https://github.com/DJI-Mobile-SDK-Tutorials/Android-VideoStreamDecodingSample/blob/master/android-videostreamdecodingsample/app/src/main/java/com/dji/videostreamdecodingsample/media/DJIVideoStreamDecoder.java), is the Mini 2 supported?

dji-dev, 2022-03-09:
Agent comment from DJI SDK in <a href='https://djisdksupport.zendesk.com/agent/tickets/61286'>Zendesk ticket #61286</a>:<br><br>尊敬的开发者，感谢您联系DJI 大疆创新
由于github不是我们主要的咨询渠道，您的问题可能跟进不及时。我们建议您通过填写表单（ https://djisdksupport.zendesk.com/hc/zh-cn/requests/new ）向我们反馈问题。或者您也可以在论坛发帖，与其它开发者交流。论坛链接：https://djisdksupport.zendesk.com/hc/zh-cn/community/topics

Dear developer, thank you for contacting DJI.
Since github is not our main consultation channel, your questions may not be followed up in time. We recommend that you fill in the form (https://djisdksupport.zendesk.com/hc/en-us/requests/new) to report problems to us. Or you can post in the forum to communicate with other developers. Forum link: https://djisdksupport.zendesk.com/hc/zh-cn/community/topics
<br><p><em><small>°°°</small></em></p>

DJI-William, 2022-03-10:
This is out of date, our latest SDK does support DJI Mini 2, we havn't update the [DJIVideoStreamDecoder.java](https://github.com/DJI-Mobile-SDK-Tutorials/Android-VideoStreamDecodingSample/blob/master/android-videostreamdecodingsample/app/src/main/java/com/dji/videostreamdecodingsample/media/DJIVideoStreamDecoder.java) to this demo. This is just a demostration of how to decoding the video stream.

Waffle1434, 2022-03-10:
> 

Thanks for the quick response. I already have a prototype app working with the latest MSDK 4.16. However, I am experiencing pretty bad `LiveFeed` latency (400 to 1000 milliseconds), so I was really hoping something like this may alleviate the video lag.

DJI-William, 2022-03-10:
DJI FLY and MSDK are using different video stream decoding code, and MSDK is very out of date so the perforamce would be worse than DJI FLY. I will inform the dev team about this issue but no guarantee because they are busy at other things.

alzin, 2022-10-08:
Is there any updates on this topic?

DJI-William, 2022-10-08:
@alzin We have added MAVIC_MINI, DJI_MINI_SE and DJI_MINI_2 to https://github.com/DJI-Mobile-SDK-Tutorials/Android-VideoStreamDecodingSample/tree/sdk_releases/4.16.2

closed 2022-10-08 by DJI-William