DJI-Mobile-SDK-Tutorials/Android-VideoStreamDecodingSample #81: SDK 4.16 - three sigsev traps on debug start

neilyoung, 2022-03-14:
With SDK 4.16 my Android Studio debugger traps repeatedly three times into SIGSEV. The module blamed is `mdecodingsample`, which somehow sounds like you. Doesn't happen with 4.15. I can proceed after each trap, but debugging is not possible, since the app terminates

[image]

[image]

[image]

`03-14 13:25:11.879 26212 26301 F libc    : Fatal signal 11 (SIGSEGV), code 1, fault addr 0x790 in tid 26301 (mdecodingsample), pid 26212 (mdecodingsample)
`

dji-dev, 2022-03-14:
Agent comment from DJI SDK in <a href='https://djisdksupport.zendesk.com/agent/tickets/61530'>Zendesk ticket #61530</a>:<br><br>尊敬的开发者，感谢您联系DJI 大疆创新
由于github不是我们主要的咨询渠道，您的问题可能跟进不及时。我们建议您通过填写表单（ https://djisdksupport.zendesk.com/hc/zh-cn/requests/new ）向我们反馈问题。或者您也可以在论坛发帖，与其它开发者交流。论坛链接：https://djisdksupport.zendesk.com/hc/zh-cn/community/topics

Dear developer, thank you for contacting DJI.
Since github is not our main consultation channel, your questions may not be followed up in time. We recommend that you fill in the form (https://djisdksupport.zendesk.com/hc/en-us/requests/new) to report problems to us. Or you can post in the forum to communicate with other developers. Forum link: https://djisdksupport.zendesk.com/hc/zh-cn/community/topics
<br><p><em><small>°°°</small></em></p>

neilyoung, 2022-03-14:
Here is all logcat grepped for "mdecodingsample" from start debugging until final termination:

```

03-14 13:36:54.622   793  3804 I ActivityManager: Force stopping com.dji.videostreamdecodingsample appid=10148 user=0: from pid 27260
03-14 13:36:54.993   793  8619 I ActivityManager: Force stopping com.dji.videostreamdecodingsample appid=10148 user=0: from pid 27269
03-14 13:36:55.080   793  8618 I ActivityManager: Force stopping com.dji.videostreamdecodingsample appid=10148 user=-1: set debug app
03-14 13:36:55.082   793  8618 I ActivityManager: START u0 {act=android.intent.action.MAIN cat=[android.intent.category.LAUNCHER] flg=0x10000000 cmp=com.dji.videostreamdecodingsample/.ConnectionActivity} from uid 2000
03-14 13:36:55.101   793  8618 I ActivityManager: Start proc 27276:com.dji.videostreamdecodingsample/u0a148 for activity com.dji.videostreamdecodingsample/.ConnectionActivity
03-14 13:36:55.210 27276 27276 W ActivityThread: Application com.dji.videostreamdecodingsample is waiting for the debugger on port 8100...
03-14 13:36:55.687   793   807 W ActivityManager: Activity pause timeout for ActivityRecord{ab37e88 u0 com.dji.videostreamdecodingsample/.ConnectionActivity t233}
03-14 13:36:58.839 11367 11367 E adbd    : failed to connect to socket 'localabstract:/com.dji.videostreamdecodingsample-0/platform-1647261412622.sock': Connection refused
03-14 13:37:05.725   793   807 W ActivityManager: Activity stop timeout for ActivityRecord{ab37e88 u0 com.dji.videostreamdecodingsample/.ConnectionActivity t233}
03-14 13:37:07.056 27276 27276 I chatty  : uid=10148(com.dji.videostreamdecodingsample) identical 7 lines
03-14 13:37:14.516 27276 27352 F libc    : Fatal signal 11 (SIGSEGV), code 1, fault addr 0x790 in tid 27352 (mdecodingsample), pid 27276 (mdecodingsample)
03-14 13:37:14.538 27359 27359 W mdecodingsample: type=1400 audit(0.0:118): avc: denied { ptrace } for scontext=u:r:untrusted_app:s0:c512,c768 tcontext=u:r:crash_dump:s0:c512,c768 tclass=process permissive=0
03-14 13:37:19.691   793  8040 I ActivityManager: Process com.dji.videostreamdecodingsample (pid 27276) has died: fore TPSL
03-14 13:37:19.723   793  8040 I ActivityManager: Start proc 27367:com.dji.videostreamdecodingsample/u0a148 for activity com.dji.videostreamdecodingsample/.ConnectionActivity
03-14 13:37:20.309   793   807 W ActivityManager: Activity pause timeout for ActivityRecord{ab37e88 u0 com.dji.videostreamdecodingsample/.ConnectionActivity t233}
03-14 13:37:20.500 27367 27394 F zygote64: java_vm_ext.cc:534] JNI DETECTED ERROR IN APPLICATION: JNI NewGlobalRef called with pending exception java.lang.UnsatisfiedLinkError: dalvik.system.PathClassLoader[DexPathList[[zip file "/data/user/0/com.dji.videostreamdecodingsample/.cache_sdk/sdkclasses2.jar", zip file "/data/user/0/com.dji.videostreamdecodingsample/.cache_sdk/sdkclasses.jar", zip file "/data/app/com.dji.videostreamdecodingsample-CU0NnwNwIU2HFvZweDeDVA==/base.apk"],nativeLibraryDirectories=[/data/app/com.dji.videostreamdecodingsample-CU0NnwNwIU2HFvZweDeDVA==/lib/arm64, /data/app/com.dji.videostreamdecodingsample-CU0NnwNwIU2HFvZweDeDVA==/base.apk!/lib/arm64-v8a, /system/lib64, /vendor/lib64]]] couldn't find "libwaes.so"
03-14 13:37:20.500 27367 27394 F zygote64: java_vm_ext.cc:534]   at void com.dji.videostreamdecodingsample.ConnectionActivity$2.run() (ConnectionActivity.java:108)
03-14 13:37:20.500 27367 27394 F zygote64: java_vm_ext.cc:534]   at com.dji.videostreamdecodingsample.ConnectionActivity$2.run(ConnectionActivity.java:108)
03-14 13:37:20.559 27367 27394 F zygote64: runtime.cc:523]   at com.dji.videostreamdecodingsample.ConnectionActivity$2.run(ConnectionActivity.java:108)
03-14 13:37:20.560 27367 27394 F zygote64: runtime.cc:523]   at com.dji.videostreamdecodingsample.ConnectionActivity.initUI(ConnectionActivity.java:276)
03-14 13:37:20.560 27367 27394 F zygote64: runtime.cc:523]   at com.dji.videostreamdecodingsample.ConnectionActivity.onCreate(ConnectionActivity.java:249)
03-14 13:37:20.562 27367 27394 F libc    : Fatal signal 6 (SIGABRT), code -6 in tid 27394 (AsyncTask #1), pid 27367 (mdecodingsample)
03-14 13:37:20.658 27397 27397 F DEBUG   : pid: 27367, tid: 27394, name: AsyncTask #1  >>> com.dji.videostreamdecodingsample <<<
03-14 13:37:20.659 27397 27397 F DEBUG   : Abort message: 'java_vm_ext.cc:534] JNI DETECTED ERROR IN APPLICATION: JNI NewGlobalRef called with pending exception java.lang.UnsatisfiedLinkError: dalvik.system.PathClassLoader[DexPathList[[zip file "/data/user/0/com.dji.videostreamdecodingsample/.cache_sdk/sdkclasses2.jar", zip file "/data/user/0/com.dji.videostreamdecodingsample/.cache_sdk/sdkclasses.jar", zip file "/data/app/com.dji.videostreamdecodingsample-CU0NnwNwIU2HFvZweDeDVA==/base.apk"],nativeLibraryDirectories=[/data/app/com.dji.videostreamde
03-14 13:37:21.034   793 27398 W ActivityManager:   Force finishing activity com.dji.videostreamdecodingsample/.ConnectionActivity
03-14 13:37:21.091   793  6696 I ActivityManager: Process com.dji.videostreamdecodingsample (pid 27367) has died: cch  CEM 

```

dji-dev, 2022-03-15:
Agent comment from William Wong in <a href='https://djisdksupport.zendesk.com/agent/tickets/61530'>Zendesk ticket #61530</a>:<br><br>Dear Client
Thank you for contacting DJI.

The sample is very old and out of date, we suggest you use the Sample Project, we will look at the video stream sample later.
Link: https://github.com/dji-sdk/Mobile-SDK-Android

Kindly Regards,
DJI Developer Support
<br><p><em><small>°°°</small></em></p>

neilyoung, 2022-03-15:
Thanks for the suggestion. Leads me to the next problem. After having compiled this monster and tried to register I'm getting this: Registration failed. I'm supposed to change the "bundle ID" (whatever this is on Android)

[image]

According to the documentation (https://developer.dji.com/mobile-sdk/documentation/quick-start/index.html#android-sample-app) I'm only supposed to add my API key (what I did). 

However, at least debugging works :/

dji-dev, 2022-03-15:
Agent comment from William Wong in <a href='https://djisdksupport.zendesk.com/agent/tickets/61530'>Zendesk ticket #61530</a>:<br><br>Check this article, make sure you have an internet access: https://sdk-forum.dji.net/hc/en-us/articles/4403040490393-How-to-register-to-MSDK-?source=search&auth_token=eyJhbGciOiJIUzI1NiJ9.eyJhY2NvdW50X2lkIjo5Nzg4MTcsInVzZXJfaWQiOjM4MTQ5NzA2NTMxNCwidGlja2V0X2lkIjo2MTUzMCwiY2hhbm5lbF9pZCI6NjMsInR5cGUiOiJTRUFSQ0giLCJleHAiOjE2NDk5Mzg5NTh9.ULl5Ua8ndUVWkS5JcmtLZxDyoGcpY9fyCdttLEynUk4
<br><p><em><small>°°°</small></em></p>

neilyoung, 2022-03-15:
OK. If you would compare the new sample app with the old VideoStreamDecodingSample - which one would come closest to the old app? I'm confused about the many samples, from which only 1/10th is working on my Mavic Pro

dji-dev, 2022-03-16:
Agent comment from William Wong in <a href='https://djisdksupport.zendesk.com/agent/tickets/61530'>Zendesk ticket #61530</a>:<br><br>The videofeederview.java file should give you a demonstration of showing video feed in Sample Project. You can also check this link to show you the image capture demo: https://github.com/dji-sdk/Mobile-SDK-Android/blob/master/Sample%20Code/app/src/main/java/com/dji/sdk/sample/demo/camera/ShootSinglePhotoView.java
<br><p><em><small>°°°</small></em></p>