DJI-Mobile-SDK-Tutorials/Android-VideoStreamDecodingSample #82: MSDK 4.16 or DJI Mini Support?

daichow, 2022-03-18:
Similar to a previous issue. I don't see the mini 1 model mentioned in [DJIVideoStreamDecoder.java](https://github.com/DJI-Mobile-SDK-Tutorials/Android-VideoStreamDecodingSample/blob/master/android-videostreamdecodingsample/app/src/main/java/com/dji/videostreamdecodingsample/media/DJIVideoStreamDecoder.java), is the Mini 1 supported?

dji-dev, 2022-03-18:
Agent comment from DJI SDK in <a href='https://djisdksupport.zendesk.com/agent/tickets/61719'>Zendesk ticket #61719</a>:<br><br>尊敬的开发者，感谢您联系DJI 大疆创新
由于github不是我们主要的咨询渠道，您的问题可能跟进不及时。我们建议您通过填写表单（ https://djisdksupport.zendesk.com/hc/zh-cn/requests/new ）向我们反馈问题。或者您也可以在论坛发帖，与其它开发者交流。论坛链接：https://djisdksupport.zendesk.com/hc/zh-cn/community/topics

Dear developer, thank you for contacting DJI.
Since github is not our main consultation channel, your questions may not be followed up in time. We recommend that you fill in the form (https://djisdksupport.zendesk.com/hc/en-us/requests/new) to report problems to us. Or you can post in the forum to communicate with other developers. Forum link: https://djisdksupport.zendesk.com/hc/zh-cn/community/topics
<br><p><em><small>°°°</small></em></p>

DJI-William, 2022-03-21:
Please have a look the issue I have replied in this thread. The decoder is an ancient code, it needs to update but currently we are wroking on MSDK 5.0 which has higher priority. It will not be updated very soon.

DJI-William, 2022-10-08:
@daichow We have added MAVIC_MINI, DJI_MINI_SE and DJI_MINI_2 to https://github.com/DJI-Mobile-SDK-Tutorials/Android-VideoStreamDecodingSample/tree/sdk_releases/4.16.2

closed 2022-10-08 by DJI-William