DJI-Mobile-SDK-Tutorials/Android-VideoStreamDecodingSample #84: Performance

gab933, 2022-04-20:
Hi,
I have few questions about performance in using different approaches:

- In this context it is better to work with Surface View or Texture View?
- Which is the advantage/disadvantage in using DJICodecManager with respect to using MediaCodec?

Thank you,

Gabriele

DJI-William, 2022-04-21:
We suggest you use SurfaceView. DJICodecManager's core is MediaCodec, so they are the same.

gab933, 2022-04-21:
Thank you for quick reply.
Therefore, which is the main differences/advatages in decoding by using **mCodecManager.sendDataToDecoder** with respect to **DJIVideoStreamDecoder.getInstance().parse** ?

DJI-William, 2022-04-21:
We don't know what is the main differences/advatages because we havn't tested and compared them.

gab933, 2022-04-21:
Thank you.
If I have some news about it I'll let you know.
Regards,

Gabriele